eclipse-upm/upm #400: Question, frequency forumla

Emaxoso, 2016-04-24:
Hi, i have a question for you:
in the header file "Buzzer.h" you define this constants,
what is the formula that you applied for calculating the value of the frequency?
#define  DO    3300    // 261 Hz
#define  RE    2930     // 294 Hz
#define  MI     2600     // 329 Hz
#define  FA     2460     // 349 Hz
#define  SOL  2190     // 392 Hz
#define  LA     1960     // 440 Hz
#define  SI      1750     // 493 Hz
_I'm using an intel edison_

Propanu, 2016-04-25:
Hi. You can use the period to to frequency formula for this: `T = 1/F`. Where T is your period in seconds and F your frequency in Hz.
There are several calculators available online that you can use for the conversion, e.g. [here](http://www.calctool.org/CALC/other/converters/freq). This [table](http://www.soundoctor.com/freq.htm) might also help.
I think initially the notes where C4 to B5 (as per labels and frequencies listed next to them) but they were shifted up F4 to E5 because the Grove Buzzer resonates best around 2300 Hz and the original labels were not changed. Will update the file when I get a chance.

Propanu, 2016-06-01:
Frequencies have been updated to match the notes defined by the header as of f850f5b7d89a2c3b8da3aaf2d35294766359dd1c. Thanks!

closed 2016-06-01 by Propanu