eclipse-upm/upm #404: zfm20:writeCmdPacket not work.

thyahan, 2016-04-26:
My project is get a raw fingerprint image from ZFM-20 Fingerprint sensor after detect a finger then sent to server by Node Js.
I need to call function inside jsump_zfm20: WriteCmdPacket to send command upload image to ZFM-20 but this function need 2 argument.

writeCmdPacket ( Uint8_t *, Number ) 

So this is my source cod.
---------------------------------
var pkt = new UintArray( [ 0x0a ] );
myFingerprintSensor.writeCmdPacket(pkt, 1);
---------------------------------

But I got this error message 

---------------------------------
myFingerprintSensor.writeCmdPacket(pkt, 1);
                    ^
Error: in method 'ZFM20_writeCmdPacket', argument 2 of type 'uint8_t *'
    at Error (native)
    at Object.<anonymous> (/root/app/app.js:28:21)
    at Module._compile (module.js:460:26)
    at Object.Module._extensions..js (module.js:478:10)
    at Module.load (module.js:355:32)
    at Function.Module._load (module.js:310:12)
    at Function.Module.runMain (module.js:501:10)
    at startup (node.js:129:16)
    at node.js:814:3
---------------------------------

I've tried to use  uint8Array function from jsupm_pn532 like this below.

---------------------------------
var pkt = new pn532.uint8Array(1);
pkt.setitem(0x0a,0);
myFingerprintSensor.writeCmdPacket(pkt, 1);
---------------------------------

And I got same error message, It would be great if you have any suggestion.

Propanu, 2016-05-02:
Hi, using the uint8Array from a different driver like that won't work because the zfm20 Node wrapper has no clue what uint8 is. Also, the first argument to _setitem()_ is the index and the second one is the value.

As you can see from the [example](../blob/master/examples/javascript/zfm20.js), the writeCmdPacket is not really needed to achieve what you mentioned. Most, if not all the functionality for this sensor is already exposed (see [zfm20 documentation](http://iotdk.intel.com/docs/master/upm/node/classes/zfm20.html)), but if you think it is useful to send command packets directly I can add the uint8 typemap to this driver.

thyahan, 2016-05-08:
I looked at the example page already. I could not find the function that I want. 
writeCmdPacket is really in need. Please do add the uint8 typemap to the driver for us. 

Thank you very in advance.

Propanu, 2016-05-11:
We've made the change (see 8e8b35273bee970e31fc8353d7065a2f57e74109). You should now be able to do something like:
```js
var pkt = new zfm20.uint8Array(1);
pkt.setitem(0, 0x0a);
myFingerprintSensor.writeCmdPacket(pkt, 1);
```
If you're using the DevKit image you can install the latest UPM ipk from our dev repo and it will include this fix: http://iotdk.intel.com/repos/3.0/intelgalactic-dev/

Propanu, 2016-06-01:
Considered fixed as of UPM version 0.7.0.

closed 2016-06-01 by Propanu

bartekwozniak93, 2016-12-10:
Hi, could you give me an example how to fetch fingerprint from sensor and save it again (eg. to another sensor) ?

Propanu, 2017-01-11:
@bartekwozniak93 if you would like to see better examples for the ZFM20 sensor, please open a new issue as we don't track issues that have been closed. Thank you!