eclipse-upm/upm #408: vcap breaking upm compilation on Edison

g-vidal, 2016-04-30:
Hi,
vcap is breaking compilation for me because there is no libjpeg.h in /usr/include and I have not found the package that provides this file :

'In file included from /home/vidal/Logiciels/upm/build/src/vcap/pyupm_vcapPYTHON_wrap.cxx:3462:0: /home/vidal/Logiciels/upm/src/vcap/./vcap.hpp:33:21: fatal error: jpeglib.h: No such file or directory #include <jpeglib.h> ^ compilation terminated. src/vcap/CMakeFiles/_pyupm_vcap.dir/build.make:62: recipe for target 'src/vcap/CMakeFiles/_pyupm_vcap.dir/pyupm_vcapPYTHON_wrap.cxx.o' failed make[2]: *** [src/vcap/CMakeFiles/_pyupm_vcap.dir/pyupm_vcapPYTHON_wrap.cxx.o] Error 1 CMakeFiles/Makefile2:4796: recipe for target 'src/vcap/CMakeFiles/_pyupm_vcap.dir/all' failed make[1]: *** [src/vcap/CMakeFiles/_pyupm_vcap.dir/all] Error 2 Makefile:123: recipe for target 'all' failed make: *** [all] Error 2
'
Can anybody please explain howto add libjpeg.h in yocto.

Many thanks

Propanu, 2016-05-02:
Easiest way would be to install `libjpeg-dev` with _opkg_. Add http://iotdk.intel.com/repos/3.0/iotdk/core2-32/ to your base feeds list in `/etc/opkg/base-feeds.conf`. Then you can do:
```
opkg update
opkg install libjpeg-dev
```
Here's a really good document that goes into more details about [Package Management on Intel Platforms](https://theiotlearninginitiative.gitbooks.io/internetofthings101/content/documentation/platform/PackageManagementSystem.html). Just keep in mind that some of the links/versions in the examples might be outdated.

I will also update this driver to use Cmake's findJPEG instead and simply not generate the target when the library isn't there.

Propanu, 2016-06-01:
The vcap driver has been updated as of UPM version 0.7.0 to check for libjpeg. If it's not found on the build system the vcap driver doesn't generate a make target.

closed 2016-06-01 by Propanu