eclipse-upm/upm #418: mpu9255 problem with javascript

teuteuguy, 2016-05-13:
Hey guys,

I went to help out a friend who is also working with an Accelerometer and the Intel Edison...

Same issue as: https://github.com/intel-iot-devkit/upm/issues/411

When trying to use the excample for mpu9250 the result is the same, fail on I2C writeReg.

The setup in this case is different. It is using a simple MPU9255 I2C board wired to a GROVE connector, connected to the Arduino GROVE breakout shield on the Intel Edison Arduino breakout board.

Any ideas ?

(I presume I may be asked to recompile the upm library again, which I have not done yet, not knowing what to change).

... :(

Propanu, 2016-05-14:
Check our list of known limitations and see if this is not one of those cases where you have to set both the Intel Edison board and the Grove Shield to 3.3V. I2C can be a bit finicky on the Arduino breakout.

Propanu, 2016-08-09:
@teuteuguy did you manage to get the sensor to work? I would mention this is not the same issue as #411 since you are using the Edison Arduino breakout instead of the Sparkfun blocks.

Propanu, 2016-08-29:
Added to list of known limitations. Also see #456, it might be possible to use the MPU9250 with an I2C bus redriver on the Intel Edison with the Arduino breakout. Will close this ticket but feel free to reference and/or continue the discussion on #456.

closed 2016-08-29 by Propanu