eclipse-upm/upm #421: Error: Illegal arguments for construction of _exports_DS18B20

josephroberts, 2016-05-19:
When I try to run the [ds18b20.js](https://github.com/intel-iot-devkit/upm/blob/master/examples/javascript/ds18b20.js) example on an Intel Edison I get the following error:

`var sensor = new sensorObj.DS18B20(0);
^

Error: Illegal arguments for construction of _exports_DS18B20
at Error (native)
at Object.<anonymous> (/usr/src/app/main.js:37:14)
at Module._compile (module.js:409:26)
at Object.Module._extensions..js (module.js:416:10)
at Module.load (module.js:343:32)
at Function.Module._load (module.js:300:12)
at Function.Module.runMain (module.js:441:10)
at startup (node.js:139:18)
at node.js:968:3`

Propanu, 2016-06-01:
Hi, are you using the latest MRAA & UPM libraries on your Edison? One wire functionality was only added with MRAA 1.0 and UPM 0.7.

Propanu, 2016-07-19:
@josephroberts are you still having this problem?

benoitma, 2016-09-12:
Same problem here with upm 0.7 and MRAA 1.3

jontrulson, 2016-09-12:
Hi, 

How are you hooking up your DS18B20(s)?  It should be a circuit like the following:
```
    -|
    U|       (D1)
    A| TX---|<--+
    R|          |
    T| RX-------o--------o  >to 1-wire data bus
    -|

               Vcc
                |
                o------------+
                |            |
                R1           |
                |            o(+)
   >1-wire------o--------o[DS18B20]
                             o(-)
                             |
                             GND

D1 = 1N4148
R1 = 4.7K
```

The error that is being reported is what you get if the constructor fails to find any devices on the 1-wire bus.  If you try the python or C++ examples, it will make this clearer.  Javascript unfortunately does not propagate the correct exception text.

benoitma, 2016-09-12:
Hi, 

I use this hardware that has a resistance pre-assembled  https://www.seeedstudio.com/One-Wire-Temperature-Sensor-p-1235.html

I figured out that this error comes when the hardware is not wired to UART. 

When wired to UART, the code is blocking when instanciating the object ```new sensorObj.DS18B20(0);```  and cannot go further. 

I will try with C++ or python examples.

jontrulson, 2016-09-12:
Did you use the diode D1 when connecting to the UART?  This is a requirement.

Also, when you say the code is blocking, do you mean it hangs?  

Can you tell me what platform you are using?  

When you see this hang, can you run 'dmesg' and see if there are any kernel errors?

benoitma, 2016-09-13:
Hi, 

I'm using an edison board with an Arduino shield  and wire the sensor to the UART slot on the Grove Starter Kit Base Shield. 

When I say the code is blocking at instanciation, I mean it doesn't hang, it stays running forever.

Propanu, 2016-09-30:
I've seen that too when the resistor value is off. Specifically, I've added an external 4.7K pull-up to a DS18B20 that already had a built in resistor with the same value (thus bringing R down to 2.35K), and the code would hang . This doesn't surprise me as timing is very strict for 1-Wire and different resistor values will alter the signal. Any chance you can measure the built in resistor for that Seeed version of this sensor?

Another thing I can recommend, build and run the MRAA [example](https://github.com/intel-iot-devkit/mraa/blob/master/src/uart_ow/uart_ow.c) for 1-Wire, as it will give you some clear error codes:
```{CPP}
MRAA_ERROR_UART_OW_SHORTED = 12,              /**< UART OW Short Circuit Detected*/
MRAA_ERROR_UART_OW_NO_DEVICES = 13,           /**< UART OW No devices detected */
MRAA_ERROR_UART_OW_DATA_ERROR = 14,           /**< UART OW Data/Bus error detected */
```

Propanu, 2016-11-07:
Here's the schematic:

[image]

Also in [Known Limitations](https://github.com/intel-iot-devkit/upm/blob/master/docs/knownlimitations.md) document.

closed 2016-11-07 by Propanu