eclipse-upm/upm #422: Mpr121 configAN3944 CDT register has invalid value

davelloper, 2016-05-19:
in  [configAN3944](https://github.com/intel-iot-devkit/upm/blob/master/src/mpr121/mpr121.cxx#L150) the CDT register is set to 0x04 which is an invalid value. Consequences are:
- lock-in of the sensor to "touched" state when touched for long periods of time ( > 10 seconds)
- lock-in sensor  to "touched" state when left alone for more than an hour

 The CDT part of register 0x5D (Highest 3 bits) should at least be set to 001 as indicated in the Application Note AN3889 , as you can see from the below screenshot of the application note:

[image]

in the current [configAN3944](https://github.com/intel-iot-devkit/upm/blob/master/src/mpr121/mpr121.cxx#L150) the highest 3 bits of 0x5D are set to 000 which is invalid.

[image]

Propanu, 2016-06-01:
Thanks for pointing this out, we will change the default value to `0x24` to match the datasheet.
Have you tried to use the `writeBytes()` function to input a valid value for the configuration register? It would be nice to know that solves this issue.

Propanu, 2016-07-19:
Fixed by b257d91702045dce8b3364f3e2ed70e213f8eeae. Thanks!

closed 2016-07-19 by Propanu