eclipse-upm/upm #424: Having trouble with import statement import pyupm

kswain1, 2016-05-25:
This is the current error when trying to import. No module named pyupm_ili9341
Maybe it is something I did wrong with the python build. Is there anyway I can specifically build the python module?

Propanu, 2016-06-01:
Hi, yes there is. When you configure the build with cmake first of all make sure `BUILDSWIG=ON` and `BUILDSWIGPYTHON=ON`. They are usually on by default. You also have the option to do Pyhon3 builds with `BUILDPYTHON3=ON` and this one is off by default. If you use command line, you can define these when you run cmake. With the curses interface (ccmake) you can simply toggle the options you need.

Once the makefiles have been generated now you can build a specific Python target by typing `make _pyupm_<name>`. In your case this would be `make _pyupm_ili9341`. If you also need to install the sensor driver you can descend into `build/src/ili9341` and simply `make install`.

Propanu, 2016-07-19:
@kswain1 were you able to build the module?

kswain1, 2016-07-19:
I was able to able to build the module. currently running into a mraa
unspecified error?

Propanu, 2016-07-20:
Do you get the error while using this module or is it unrelated?

kswain1, 2016-07-20:
Yes it is related to the module. When I run an example program I am
receiving this error. Will try deleted and redownloading the code.
Currently followed this module to build the process
https://learn.sparkfun.com/tutorials/using-an-lcd-on-the-edison.

Thank you for your time!

Propanu, 2016-08-02:
By the way you can also use `opkg` to update/install our libraries from the [IOTDK repository](http://iotdk.intel.com/repos/3.5/intelgalactic/opkg/i586/). It will save you the hassle of a build and the ili9341 module is there in our latest library. You can find more on this in the [Intel® Edison Board User Guide](https://software.intel.com/en-us/node/623543).

Propanu, 2016-08-09:
Considering this resolved for now. Let us know if you have more questions!

closed 2016-08-09 by Propanu