eclipse-upm/upm #427: Proposal to group all python upm modules in  a folder upm

g-vidal, 2016-06-10:
Hello,

This is just a proposal to produce cleaner output. 
upm has grown **a lot** and now if one compiles the ful library the site-packages (or dist-packages for Deb users) get crowded with hundreds of files. I feel that it might be preferable to have one folder containing all modules.
Of course this implies that in the programs we will have to use 
`from upm import pypum_module`
I can do that with my configuration file but I would propose that it becomes default.  I cannot see what might be the unwanted consequences...and this could be a little too much tidy.

Thanks for the enormous work!
best regards

Propanu, 2016-06-14:
Not a bad idea at all. We are aware that the UPM library is now quite big, thus we started looking into split packages that would allow users to install the drivers only for the language their are interested in. Once that is complete, we can look into grouping the sensors a little better too.
Definitely something to consider, thanks for bringing it up!

Propanu, 2016-10-12:
Will be added with #468, but we still need to clarify the change in our documentation pages.

pylbert, 2016-10-13:
Fixed in #468, updated changelog.md to reflect the new syntax for loading python modules.

closed 2016-10-13 by pylbert