eclipse-upm/upm #432: Change Values for Temp Sensor v1.2

madhatter215, 2016-06-16:
Request to support version 1.2 of the temperature sensor. 

Grove Temp Sensor v1.2
============================
value of the thermistor = 4275
value of resister = 100000 (or 100k)

Propanu, 2016-08-09:
This was updated and tested as of UPM v0.7.3. The new constructor also allows you to use different values for B and R0 for compatibility with other (or older) parts. Thanks for pointing out the change in specs!

closed 2016-08-09 by Propanu

deadprogram, 2016-08-11:
Seems like the values being returned by the 1.2 sensor are perhaps degrees C * 2. Possibly related to Firmata analog value?

Propanu, 2016-08-11:
Heh, yes, the ADC on the Arduino 101 uses 3.3V as reference. When using a Grove shield set to 5V you need to scale the values back down by 0.66. The second parameter of the constructor allows you to do so.

deadprogram, 2016-08-11:
Yes, because even changing the third param to 3.3 still same error:

```
// analog voltage, usually 3.3 or 5.0
var OTP538U_AREF = 3.3;
```