eclipse-upm/upm #434: how to compile upm on ubuntu 14.04

ouki-wang, 2016-06-22:
I compile upm using command "cmake  -DBUILDSWIGPYTHON=OFF -DCMAKE_INSTALL_PREFIX:PATH=../_install ../"

-- checking for module 'libbacnet'
--   package 'libbacnet' not found
-- checking for module 'libbacnet'
--   package 'libbacnet' not found

i have compiled bacnet stack 0.8.3 but I don't known how to install libbacnet。

Propanu, 2016-06-23:
Hi, bacnet is not a hard dependency. If it's installed on your system it will build those sensor drivers too but otherwise they'll be skipped. Same thing for ozw and modbus drivers. Are you sure this is what causes your build to fail?

ouki-wang, 2016-06-24:
Not bacnet library compilation errors caused, but caused -lpthreads
how to solve the problem?
@Propanu tks for your help

Run Build Command:/usr/bin/make "cmTryCompileExec3773774936/fast"
/usr/bin/make -f CMakeFiles/cmTryCompileExec3773774936.dir/build.make CMakeFiles/cmTryCompileExec3773774936.dir/build
make[1]: Entering directory `/home/dfrobot/Desktop/upm/build/CMakeFiles/CMakeTmp'
/usr/bin/cmake -E cmake_progress_report /home/dfrobot/Desktop/upm/build/CMakeFiles/CMakeTmp/CMakeFiles 1
Building C object CMakeFiles/cmTryCompileExec3773774936.dir/CheckFunctionExists.c.o
/usr/bin/cc   -DCHECK_FUNCTION_EXISTS=pthread_create   -o CMakeFiles/cmTryCompileExec3773774936.dir/CheckFunctionExists.c.o   -c /usr/share/cmake-2.8/Modules/CheckFunctionExists.c
Linking C executable cmTryCompileExec3773774936
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTryCompileExec3773774936.dir/link.txt --verbose=1
/usr/bin/cc   -DCHECK_FUNCTION_EXISTS=pthread_create    CMakeFiles/cmTryCompileExec3773774936.dir/CheckFunctionExists.c.o  -o cmTryCompileExec3773774936 -rdynamic -lpthreads 
/usr/bin/ld: cannot find -lpthreads
make[1]: Leaving directory `/home/dfrobot/Desktop/upm/build/CMakeFiles/CMakeTmp'
collect2: error: ld returned 1 exit status
make[1]: *** [cmTryCompileExec3773774936] Error 1
make: *** [cmTryCompileExec3773774936/fast] Error 2

jontrulson, 2016-06-24:
Hi,

Your log doesn't seem to have anything to do with UPM or bacnet...  What are you building on?  What are you trying to do?

ouki-wang, 2016-06-24:
i want to run upm on ubuntu and  control arduino via firmata protocol

jontrulson, 2016-06-24:
Looking at the above errors in more detail - these look like the normal checks cmake does when looking for a thread library.  It will look for both libpthread, and libpthreads.  It will look for these libraries by attempting to build a small test program.  

Having it fail to find libpthreads would be normal, since Linux uses libpthread instead.  

However, it should not fail the build of UPM, assuming libpthread was found.  So there is something else wrong here, and there is not enough information to determine what...

Propanu, 2016-07-19:
@ouki-wang did you resolve the issue?

Propanu, 2016-08-09:
Considered resolved. As a note, you can also check our [travis.yml](../blob/master/.travis.yml) to see how we setup our Ubuntu builds for CI. If you cloned our repository this same file will allow you to build our library and follow the steps with [Travis CI](https://travis-ci.org).

closed 2016-08-09 by Propanu