eclipse-upm/upm #440: Error:upm::Jhd1313m1' must implement the inherited pure virtual method 'upm::LCD::write

undersky, 2016-07-17:
I am using Intel Edison eclipse IDE. And I import an Iot Example "Display temperature on LCD". After build, I got error:
> Description	Resource	Path	Location	Type
> The type 'upm::Jhd1313m1' must implement the inherited pure virtual method 'upm::LCD::write' 	> testlcd.cpp	/testlcd/src	line 164	Code Analysis Problem

And the error line is : `upm::Jhd1313m1 *lcd = new upm::Jhd1313m1(0, 0x3E, 0x62);`

Propanu, 2016-07-18:
Hi! It's an issue with Codan (the Code Analyzer) not refreshing after the libraries were updated. The code will build and run fine, but to get rid of the error message you have to right click on the project in Eclipse and select `Index -> Freshen All Files`.

undersky, 2016-07-19:
Thanks @Propanu 

I have solve this problem by your advise. 

**You are so helpful.**

closed 2016-07-19 by undersky