eclipse-upm/upm #444: UPM Runtime Error: MPL3115A2: incorrect device id

youcangetme, 2016-07-28:
I updated my projects to the latest OS for Edison and did a opkg and saw a bunch of stuff for upm come across.  0.7.2 if memory serves me.

Anyways, when I rebooted all of my devices are now showing the following and I'm in big trouble. Not sure what is going on here. The app hasn't changed just the OS and updated the packages;

`Caused by: java.lang.RuntimeException: UPM Runtime Error: MPL3115A2: incorrect device id
        at upm_mpl3115a2.javaupm_mpl3115a2JNI.new_MPL3115A2__SWIG_2(Native Method)
        at upm_mpl3115a2.MPL3115A2.<init>(MPL3115A2.java:47)
`

youcangetme, 2016-07-29:
The bug is not in the upm package. Closing this ticket...

closed 2016-07-29 by youcangetme

Propanu, 2016-07-29:
I was about to say it looks like the sensor doesn't get detected on the I2C bus, in which case there's not much you can do from a UPM perspective. Did you figure out what is causing it?