eclipse-upm/upm #445: Cannot run bno055.js example on Intel Edison Yocto 3.5

tripish, 2016-07-29:
Upon running the example with Node, get the following error.  Noting that the Intel Edison only exposes I2C bus 6, I also tried modifying the example so the constructor was BNO055(6, 0x28), with the same error.

`node bno055.js 
/home/root/upm/examples/javascript/bno055.js:34
var sensor = new sensorObj.BNO055();
             ^

Error: Illegal arguments for construction of _exports_BNO055
    at Error (native)
    at Object.<anonymous> (/home/root/upm/examples/javascript/bno055.js:34:14)
    at Module._compile (module.js:409:26)
    at Object.Module._extensions..js (module.js:416:10)
    at Module.load (module.js:343:32)
    at Function.Module._load (module.js:300:12)
    at Function.Module.runMain (module.js:441:10)
    at startup (node.js:139:18)
    at node.js:968:3`

On a fresh install of:

- Intel provided Yocto Linux 3.5
- mraa 1.1.2
- upm 0.7.2
- node 4.4.3

tripish, 2016-07-29:
Upon further investigation...

```
root@edison:~/upm/examples/javascript# journalctl -f
-- Logs begin at Sat 2000-01-01 00:00:11 UTC. --
Jul 29 18:43:51 edison libmraa[1307]: edison: Arduino board not detected, assuming Intel Edison Miniboard
Jul 29 18:43:51 edison libmraa[1307]: libmraa initialised for platform 'Intel Edison' of type 2
Jul 29 18:43:51 edison libmraa[1307]: i2c_init: Selected bus 6
Jul 29 18:43:51 edison libmraa[1307]: i2c6: write_byte_data: Access error: Remote I/O error
Jul 29 18:44:00 edison libmraa[1314]: libmraa version v1.1.2 initialised by user 'root' with EUID 0
Jul 29 18:44:00 edison libmraa[1314]: gpio214: init: Failed to write to 'export': No such device
Jul 29 18:44:00 edison libmraa[1314]: edison: Arduino board not detected, assuming Intel Edison Miniboard
Jul 29 18:44:00 edison libmraa[1314]: libmraa initialised for platform 'Intel Edison' of type 2
Jul 29 18:44:00 edison libmraa[1314]: i2c_init: Selected bus 6
Jul 29 18:44:00 edison libmraa[1314]: i2c6: write_byte_data: Access error: Remote I/O error
```

closed 2016-07-29 by tripish

shantanoo-desai, 2017-06-20:
@tripish Hello I seem to be facing the same problem with the Python example for bno055. What are the recommended steps to resolve the issue?

alext-mkrs, 2017-06-22:
@tripish, the log looks like if you were running it on miniboard, is that correct? The log also looks like there's a HW-level problem when it tries to send the data, are you sure you have proper level converters (miniboard exposes pins at native 1.8V, unlike 3.3/5 on Edison Board for Arduino)/pull-ups for your specific setup?

shantanoo-desai, 2017-06-22:
@alext-mkrs what kind of level converters would you suggest ?

alext-mkrs, 2017-06-24:
I haven't used the miniboard-based setup that much, so I don't have any first-hand level converter exeprience. However a while ago, when planning some project, which I haven't implemented, I've done some research and, for example, I bought a couple of [these](https://www.sparkfun.com/products/12009). That's by no means to say this model is "the best", just the one I had available at the time and it was reported to work with Edison's miniboard. You better do some research, specific to your project, as setup details may affect the choice.