eclipse-upm/upm #446: upm compilation crashes on BME280

g-vidal, 2016-08-02:
Hi,
I cannot compile upm from git sources from here here is the error message :

```
Scanning dependencies of target _pyupm_bme280
[  8%] Building CXX object src/bme280/CMakeFiles/_pyupm_bme280.dir/pyupm_bme280PYTHON_wrap.cxx.o
/home/vidal/Logiciels/upm/build/src/bme280/pyupm_bme280PYTHON_wrap.cxx: In function 'PyObject* _wrap_new_BME280__SWIG_0(PyObject*, PyObject*)':
/home/vidal/Logiciels/upm/build/src/bme280/pyupm_bme280PYTHON_wrap.cxx:3387:56: error: invalid new-expression of abstract class type 'upm::BME280'
       result = (upm::BME280 *)new upm::BME280(arg1,arg2);
                                                        ^
In file included from /home/vidal/Logiciels/upm/build/src/bme280/pyupm_bme280PYTHON_wrap.cxx:3324:0:
/home/vidal/Logiciels/upm/src/bme280/./bme280.hpp:22:7: note:   because the following virtual functions are pure within 'upm::BME280':
 class BME280 : public ITemperatureSensor, public IHumiditySensor, public IPressureSensor {
       ^
In file included from /home/vidal/Logiciels/upm/src/bme280/./bme280.hpp:3:0,
                 from /home/vidal/Logiciels/upm/build/src/bme280/pyupm_bme280PYTHON_wrap.cxx:3324:
/home/vidal/Logiciels/upm/src/bme280/../upm/iTemperatureSensor.hpp:37:20: note: 	virtual int upm::ITemperatureSensor::getTemperatureCelsius()
        virtual int getTemperatureCelsius () = 0;
                    ^
/home/vidal/Logiciels/upm/build/src/bme280/pyupm_bme280PYTHON_wrap.cxx: In function 'PyObject* _wrap_new_BME280__SWIG_1(PyObject*, PyObject*)':
/home/vidal/Logiciels/upm/build/src/bme280/pyupm_bme280PYTHON_wrap.cxx:3464:51: error: invalid new-expression of abstract class type 'upm::BME280'
       result = (upm::BME280 *)new upm::BME280(arg1);
                                                   ^
src/bme280/CMakeFiles/_pyupm_bme280.dir/build.make:70: recipe for target 'src/bme280/CMakeFiles/_pyupm_bme280.dir/pyupm_bme280PYTHON_wrap.cxx.o' failed
make[2]: *** [src/bme280/CMakeFiles/_pyupm_bme280.dir/pyupm_bme280PYTHON_wrap.cxx.o] Error 1
CMakeFiles/Makefile2:1832: recipe for target 'src/bme280/CMakeFiles/_pyupm_bme280.dir/all' failed
make[1]: *** [src/bme280/CMakeFiles/_pyupm_bme280.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
```
Did I do something wrong ? Is there a patch somewhere.
I tried to compile by myself because there was no puypm_bme280 in the default library installed.

Thanks for your help.

Propanu, 2016-08-02:
HI @g-vidal! The BME280 driver has been merged into BMP280 since 6241e20dda9cf52e03272c61e2496d2e9c3e49c0 and should build without any issues. If you still have the old target you'll want to clean the build directory and regenerate the makefiles.

g-vidal, 2016-08-02:
Thanks for your quick answer but there must be something wrong....
BMP280 is the successor of BMP180 and both are
pressure temperature sensors
meanwhile
BME280 is a Pressure Temperature HUMIDITY sensor... 
It is not possible to merge ifferent sensors and on the website it seems clear that BME280 inherits functions from BMP280 and has one more on top of it 
http://iotdk.intel.com/docs/master/upm/classupm_1_1_b_m_e280.html

Am I wrong?

best

Propanu, 2016-08-02:
The BME280 is a BMP280 with extra features (humidity) just like you mentioned, but the pressure and temperature parts are common. That's why it derives from the BMP280 class. It just so happens that we named the module BMP280, but once you import the module you can use either one of the two sensors.

Using it should be straightforward as shown in the [example](https://github.com/intel-iot-devkit/upm/blob/master/examples/python/bme280.py).

g-vidal, 2016-08-02:
OK thanks I get it now
So what is the  the solution get a successful compilation?
Fully remove the bme280 directory from upm/src ... is that right ?
I'll check!
Thanks  and sorry to be slow at understanding  ;(
best

Propanu, 2016-08-02:
If it's still there yes, remove it. See 84144d0c8fd66e90af21e5ef8a804dc08839bb0f.

g-vidal, 2016-08-02:
Ok 
I have understtod the problem. I have removed my upm version and clone it now. In this fresh version there is no BME280 directory. before compiling I did a git pull  but something must have gone wrong because the  directory was not removed.. That's why the compilation failed on BME.

Thanks again for your patience, it is compiling now ....

closed 2016-08-02 by g-vidal

Propanu, 2016-08-02:
You're welcome. Glad you have it building again!