eclipse-upm/upm #447: Error: Syntax error in input(3)

vitcozzolino, 2016-08-04:
Hi,
I'm trying to compile and build UPM following the suggested steps on my Debian 32-bit machine. Apparently there are some issues with SWIG (I guess). In multiple cases, after launching the `make `command, I receive the following error (for some of the driver libraries):

```
[ 57%] Swig source
/home/osboxes/building/upm/src/bmx055/bma250e.hpp:84: Error: Syntax error in input(3).
src/bmx055/CMakeFiles/_pyupm_bmx055.dir/build.make:53: recipe for target 'src/bmx055/pyupm_bmx055PYTHON_wrap.cxx' failed
make[2]: *** [src/bmx055/pyupm_bmx055PYTHON_wrap.cxx] Error 1
CMakeFiles/Makefile2:9293: recipe for target 'src/bmx055/CMakeFiles/_pyupm_bmx055.dir/all' failed
make[1]: *** [src/bmx055/CMakeFiles/_pyupm_bmx055.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
```

This happens also for bno055 and others. Actually the error can be solved if, in the affected files, the following string `typedef enum : uint8_t {` is changed into `typedef enum {`. But I'm pretty sure that's not the right way to do it.

Any suggestions?

jontrulson, 2016-08-04:
This appears to be an issue with your C++ compiler not supporting the C++11 standard.  You should try to use a compiler of at least version (gcc) 4.8 or 4.9.  Several of the current drivers make use of C++11 features.

vitcozzolino, 2016-08-05:
Thanks for the reply. I'm actually using version 4.9.2.

```
root@osboxes:/home/osboxes/building# gcc --version
gcc (Debian 4.9.2-10) 4.9.2
Copyright (C) 2014 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

jontrulson, 2016-08-05:
Hi,  can you describe how you are building UPM?  When you run cmake, it should tell you whether C++11 was detected and what option was used to enable it.  What version of SWIG are you using?  I'm using gcc/g++ 4.9 and SWIG 3.0.6 and this compiles without problem.

vitcozzolino, 2016-08-08:
Hi,
I've followed the instruction to install UPM showed on the git repo (https://github.com/intel-iot-devkit/upm/blob/master/docs/building.md). When I launch `cmake `inside the `build ` folder that's the output:

```
root@osboxes:/home/user/upm/build# cmake .. -DBUILDSWIGNODE=OFF
-- The C compiler identification is GNU 4.9.2
-- The CXX compiler identification is GNU 4.9.2
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Looking for include file pthread.h
-- Looking for include file pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.28") 
-- checking for module 'mraa>=1.1.1'
--   found mraa, version 1.2.1
INFO found mraa version: 1.2.1
INFO - UPM Version v0.7.2-15-g49e0459
-- Found SWIG: /usr/bin/swig2.0 (found version "2.0.12") 
INFO - Target arch is i586
-- Performing Test COMPILER_SUPPORTS_CXX11
-- Performing Test COMPILER_SUPPORTS_CXX11 - Success
-- Performing Test COMPILER_SUPPORTS_CXX0X
-- Performing Test COMPILER_SUPPORTS_CXX0X - Success
-- Found PythonInterp: /usr/bin/python2.7 (found suitable version "2.7.9", minimum required is "2.7") 
-- Found PythonLibs: /usr/lib/i386-linux-gnu/libpython2.7.so (found suitable version "2.7.9", minimum required is "2.7") 
INFO Generating empty /home/user/upm/build/src/pyupm_doxy2swig.i
-- checking for module 'libmodbus'
--   package 'libmodbus' not found
-- checking for module 'libbacnet'
--   package 'libbacnet' not found
-- checking for module 'libmodbus'
--   package 'libmodbus' not found
-- checking for module 'libbacnet'
--   package 'libbacnet' not found
-- checking for module 'libbacnet'
--   package 'libbacnet' not found
-- checking for module 'libmodbus'
--   package 'libmodbus' not found
-- Looking for mraa_firmata_init in mraa
-- Looking for mraa_firmata_init in mraa - not found
-- checking for module 'libopenzwave'
--   package 'libopenzwave' not found
-- checking for module 'libbacnet'
--   package 'libbacnet' not found
-- Could NOT find JPEG (missing:  JPEG_LIBRARY JPEG_INCLUDE_DIR) 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/user/upm/build
```

jontrulson, 2016-08-08:
On 08/08/2016 05:55 AM, Vittorio Cozzolino wrote:
> Hi,
> I've followed the instruction to install UPM showed on the git repo
> (https://github.com/intel-iot-devkit/upm/blob/master/docs/building.md). When
> I launch |cmake|inside the |build| folder [...]

This is most likely the problem - C++11 support was not added to SWIG 
until release 3.0.  I use 3.0.6.  If you do not intend to use the 
Javascript, Python, or Java bindings, you can disable SWIG by adding 
"-DBUILDSWIG=OFF -DBUILDSWIGPYTHON=OFF -DBUILDSWIGNODE=OFF 
-DBUILDSWIGJAVA=OFF" to your cmake command line.

If you need SWIG bindings, then you will need to upgrade your Swig to at 
least 3.0.5 (for NODE).  You may have to build it yourself.

vitcozzolino, 2016-08-09:
Ok I've compiled with the following flags "-DBUILDSWIG=OFF -DBUILDSWIGNODE=OFF" and it works (I need the Python bindings). I will explore a bit further the issue and make a comparison with other build environments I use. Meanwhile, thanks for the help!

Propanu, 2016-08-09:
We're glad you could get it to work. Feel free to reopen this or start a new issue if you have other questions!

closed 2016-08-09 by Propanu

shilendras, 2017-04-21:
[image]

This is the issue I am having. I am not able to build upm

pylbert, 2017-04-21:
@shilendras, make sure you have built/installed mraa 1.6.1.  This is the latest version - it's possible you have an older mraa installed that doesn't meet the version requirement.