eclipse-upm/upm #448: about library of led apa102

KaboChueng, 2016-08-05:
Hello!
I don't know how to use the function - setLeds(). I try to pass an array as the third parameter,but it's wrong. Can you make an example?

KaboChueng, 2016-08-05:
Hello!
Another big problem. Sometimes, program would  get stuck and couldn't stop. 
Is something wrong with signal transmission?
We really need your help!

windmillknight, 2016-08-08:
I have the problem too . I write a piece of code like this:
========================================================================
 function go (i,end){
     if(i<end){
       if(i !== 0){
           var j = (i-1);
         ledStrip.setLed(j,1,0,0,0);      
       }
     ledStrip.setLed(i,1,255,255,255);
     ledStrip.pushState();
     setTimeout(go,500,++i,end);
     }
     else{
          ledStrip.setAllLeds(1,0,0,0);
          return ;
     } 
 }
var lib = require('jsupm_apa102');
var ledStrip = new lib.APA102(30,0,true);
go(0,30);
======================================================================
when I run this for some times ,program would get stuck,
and I have to reboot my edison .Why?

Propanu, 2016-08-11:
Hi, I think I have one of these LED strips and I will take a look to see what can be done. Meanwhile, there's a known issue with SPI on Edison where you have to turn off power management for most of the SPI sensor modules to work correctly. You can do so with:
```
echo on > /sys/devices/pci0000\:00/0000\:00\:07.1/power/control
```

windmillknight, 2016-08-11:
Hello. This problem has been solved after I reload my edison system.It works currectly now ! Thank you very much ! And can you solve my another quetion about mpu9150 ?

Propanu, 2016-08-26:
@KaboChueng, setLeds() has two overloads:
```
setLeds(uint16_t startIdx, uint16_t endIdx, uint8_t brightness, uint8_t r, uint8_t g, uint8_t b)
```
which should be straightforward to use and
```
setLeds(uint16_t startIdx, uint16_t endIdx, uint8_t* colors)
```
If you want to call the second form in Node.js or Pyhon you'll need to use the `uint8Array()` wrapper for proper type mapping. Check our pn532 examples on how to use that. From Java you should be able to use byte[] as your array.

closed 2016-08-30 by Propanu