eclipse-upm/upm #449: Something Wrong with MPU9150 moudle!

windmillknight, 2016-08-10:
Hello! I faced a trouble whith upm's Mpu9150 moudle.

My Mpu9150  is like this 
[image]
MPU-9150 GY-9150 

--------------------------------------------------------------------------------------------------------
and I install upm moudle by 
"obkg install upm"  
from 
"src mraa-upm http://iotdk.intel.com/repos/3.5/intelgalactic/opkg/i586"

Is that right?
_________________________________________________________________________

sensor.init();
       ^
Error: UPM Invalid Argument: Unknown error in I2c::readReg()
    at Error (native)
    at Object.<anonymous> (/home/root/intel-iot-services-orchestration-layer-master/node_modules/hope-demo/startkit/thing_bundle/new_things_by_user/MPU9150/输出/kernel.js:7:8)
    at Module._compile (module.js:409:26)
    at Object.Module._extensions..js (module.js:416:10)
    at Module.load (module.js:343:32)
    at Function.Module._load (module.js:300:12)
    at Function.Module.runMain (module.js:441:10)
    at startup (node.js:139:18)
    at node.js:968:3
_
___________________________________________________________________________
 Why this happen? please help me
my edison version is v3.10
mraa version is v1.2.2

jontrulson, 2016-08-11:
Hi,

Some of these sensors just do not play well with Edison I2C on an Arduino breakout.  Check the system logs (journalctl) while trying to use the sensor and see if you are getting I2C read/write errors reported.

Unfortunately, if that is the case, we can't do anything about it.

windmillknight, 2016-08-12:
Aug 12 03:41:54 Cheese libmraa[593]: [[1;39medison: Arduino board detected[[0m
Aug 12 03:41:54 Cheese libmraa[593]: [[1;39mlibmraa initialised for platform 'I
ntel Edison' of type 2[[0m
Aug 12 03:41:54 Cheese libmraa[593]: [[1;39mi2c_init: Selected bus 0[[0m
Aug 12 03:41:54 Cheese libmraa[593]: [[1;31mInvalid i2c bus 0, moving to defaul
t i2c bus 6[[0m
Aug 12 03:41:54 Cheese libmraa[593]: [[1;31mi2c6: read_byte_data: Access error:
 Remote I/O error[[0m

jontrulson, 2016-08-12:
Yes, that's the error.  One thing you might try is to use a voltage of 3.3v.  That helps with some sensors that show this problem with Edison/Arduino.  Also, if you have any other devices on the I2C bus, you might try to remove them and see if that helps.

Propanu, 2016-08-23:
Hi @windmillknight. On the Edison with the Arduino breakout this sensor is only partially usable and it has to be alone on the I2C bus. You'll need to drop the board to 3V3 using jumper J9 and only the accelerometer, gyro & temperature will be available. In other words the built-in AK8975 compass doesn't get detected. You can either use the MPU60x0 class as is or the MPU9150 with the 4th constructor parameter `enableAk8975` set to false.

Propanu, 2016-11-07:
I'll close here, same topic in #456 which we can use to continue the discussion if needed.

closed 2016-11-07 by Propanu