eclipse-upm/upm #450: OTP538u error "object temperature out of range" on Firmata subplatform

deadprogram, 2016-08-11:
When trying to run the `examples/javascript/otp538u.js` program using the Firmata subplatform, I receive the following error:

```
                roundNum(tempIRSensor_obj.objectTemperature(), 2) +
                                          ^

Error: UPM Out of Range: objectTemperature: object temperature out of range.
    at Error (native)
    at checkTemp [as _repeat] (/root/test/temp.js:44:29)
    at wrapper [as _onTimeout] (timers.js:275:11)
    at Timer.listOnTimeout (timers.js:92:15)
```

Propanu, 2016-08-11:
I bet this is the same issue as in #432. The raw values from the ADC need to be scaled down, thus I will add the option to this driver too. We are actually working towards a generic analog class that will allow setting a scale & offset for calibration and all analog sensors will be modified to incorporate this.

deadprogram, 2016-08-11:
Yes, because even changing the third param to 3.3 results in same error:

```
// analog voltage, usually 3.3 or 5.0
var OTP538U_AREF = 3.3;
```

deadprogram, 2016-08-11:
Also, I tried switching the voltage on the shield to 3.3V, but same error.

Propanu, 2016-08-11:
@jontrulson any thoughts, are the hard-coded coefficients for this sensor potentially causing this? I recall this sensor was incompatible with the G2 and had to be 'customized' to the Edison to some extent.

jontrulson, 2016-08-11:
I remember it being a difficult sensor to work on :)  The issue with G2 was the 'usual' issues I reported with some analog sensors not working well (or at all).  On edison, the sensor worked - I don't recall having to do anything special.  It required a nominal distance from the target of 9cm.  

It works my measuring ambient temperature via a thermistor and a lookup table (RT, indexed by computed resistance), then, measuring the thermopile voltage, and with that looking it up in another table (indexed by computed ambient temp and thermopile voltage, IIRC).   

The ambient temperature seems to at least be in range or you would be seeing a different exception.  So "something" in the thermopile analog measurement itself or the voltage computation is malfunctioning, since an appropriate range is never found in the VT (thermopile) table, hence the exception.  It thinks the object temperature is either below -20C or above 200C, which seems unlikely, or (more likely) the voltage it's searching for in the table is just really wrong to start with.

As to what that "something" is, debugging would be needed to find out.  Nothing leaps out at me.  First thing to do is get the ambient temperature and maybe work from there.  Does the ambient temperature seem to be reasonable?

Maybe find out what is being computed for 'voltage' at line 180, then look in thermopile_vt_table.hpp and see what is going on.

jontrulson, 2016-08-11:
Looking at the constructor - I do see a potential problem.  And yes, @Propanu, you were correct -- for the Edison we did need to insist on an ADC with a 12 bit resolution, which was not possible on the G2.  That could be why I didn't log it as a problem on G2.

I believe the original Seeed code was using a 1.2v analog ref voltage to get around the 1024 limit of the arduino, since the sensor required 1mV resolution.  Does this firmata platform support 12b ADC resolution?  Maybe something else to look at...

Propanu, 2016-08-11:
No, Firmata is running on an Arduino 101 and thus is limited to 10-bit resolution only for Arduino IDE compatibility. Would this be a quick fix? To add a line that checks whether we're on a subplatform and then set the resolution in the calculations accordingly?

jontrulson, 2016-08-11:
Well, I'm not sure... It would seem to me that if this is the problem, it would still be working (no exceptions thrown), it just would not be very accurate.

It would be nice to be able to query MRAA to find out the highest available ADC resolution supported - but as you can see right now it is just setting 12b unconditionally, since IIRC mraa_aio_set_bit() is just a scaler and it doesn't care whether the resolution is actually supported by the HW.

jontrulson, 2016-08-16:
Ok, I've committed some changes to the driver that should allow it to work on the 101 via firmata.  There were a couple of issues - the main one being that setting an ADC resolution beyond 10b on the 101 yields nonsensical results.  In addition, after the first ADC read, there seems to be a time period where all ADC reads just return 0, so I've added a hack to the ctor to handle that case - read from both ADC's, discard the results, and sleep for half a second before continuing operation.

Be sure to power the sensor via the 3.3v pin, and make sure to select 3.3v as the aref in the constructor.  The commit is (dc93fb11ff761a8a1b3302c54d9787e93ec663ec).  

Good luck.

Propanu, 2016-08-30:
@deadprogram did the fix work for you?

deadprogram, 2016-10-21:
Hi @Propanu just wondering when version 0.8+ of UPM is going to go live on the 01org repos so the gateways will have this fix available. Thanks!

Propanu, 2016-10-26:
Hi @deadprogram, the RPMs should be available: https://download.01.org/iotgateway/. Let me know if I can close this. Thanks!

deadprogram, 2016-10-31:
This is now working, thanks very much!

Propanu, 2016-11-03:
Thanks for the confirmation!

closed 2016-11-03 by Propanu