eclipse-upm/upm #452: Backlight on grove lcd RGB backlight v2.0

edwardsp, 2016-08-16:
The backlightOff function doesn't work:

```
var LCD = require('jsupm_i2clcd');
var myLcd = new LCD.Jhd1313m1(0);
myLcd.backlightOff();
```

jontrulson, 2016-08-16:
Looking at the code, it seems whomever added the backlight support code did it improperly.  The backlight*() methods should NOT be calling expandWrite() directly.  I am not surprised it fails.

This will need investigation.

Propanu, 2016-08-19:
The backlight on/off functions were added for lcm1602 LCDs, not for JHD1313. Guess I'm to blame here for not documenting it enough. Also, for completeness, the JHD1313 class should override them accordingly. Will look into it when I get the chance.

jontrulson, 2016-08-19:
The main issue I saw was calling expandWrite() - It should probably be command() - this way it would work on gpio and expander use.  The Grove LCD may have some other completely different way of controlling the backlight.  Of course I am assuming that one of the control GPIO's is used to control it.

But I don't know anything about how backlight is implemented on these devices, I would need to investigate to be sure how it should actually work.  I am pretty certain that calling expandWrite() there can't be right :)

Propanu, 2016-08-22:
I think I used expandWrite() because that was the way we were setting it in the constructor. Also, not all LCDs support that feature. I think it's just one bit that you can set on/off and the brightness level would be controlled from a potentiometer on the LCD. I'll follow your suggestion to see if it can be done with command() instead.

For the Grove RGB LCD, it can be done with simple calls to the setColor() function.

edwardsp, 2016-08-23:
I had a look and have found a way to control the backlight.

backlight on: m_i2c_lcd_rgb.writeReg(0x08, 0xAA);
backlight off: m_i2c_lcd_rgb.writeReg(0x08, 0x00); 

This approach is slightly better than setting the color to (0,0,0) as it will remember the last color when the backlight is turned back on.

Propanu, 2016-08-31:
@edwardsp followed your suggestion and implemented the functions for the Grove RGB LCD too, in addition to a bit of code cleanup. See 53041928135de2c6c949c3694f30dcf5b12e62bb. Thanks a lot for the suggestion!

@jontrulson regarding LCM1602, it is possible to control GPIO LCDs that have backlight too, but that sort of extension will likely require a new constructor with 2 extra pins in there + some logic added for the 2 backlight functions. Since the backpack version is likely used more, I will leave it as is for now unless someone opens up a request.

closed 2016-08-31 by Propanu

jontrulson, 2016-08-31:
@Propanu It's probably ok, as expandWrite() will make sure you are actually on an I2C device before ruining your day :)