eclipse-upm/upm #453: grove LCD module still using really bad i2c commands

arfoll, 2016-08-16:
LCD module is not using full i2c commands and instead is relying on usleep and other methods to throw tiny segments of data down the i2c line. This is causing a bunch of reliability issues.

For example:
https://github.com/intel-iot-devkit/upm/blob/master/src/lcd/lcm1602.cxx#L396
this calls write4bits() twice in i2c for no good reason. This can be done in much fewer i2c commands resulting in much better reliability and speed.

jontrulson, 2016-08-16:
Hi,

I am confused a little bit.  Are you talking about the Grove RGB LCD display (JHD1313M1) specifically or just a generic LCM1602?

For the code you linked to, this is actually the only way to communicate with this particular device.  The "plain' LCM1602 device is actually fully GPIO controlled, with control pins (4 IIRC) and 8 data lines.   

Almost no one uses it in this manner since it would require 12 GPIO's to operate.  So, the device also supports a 4-bit mode, which only requires 8 GPIO's - 4 for control, 4 for data.  And yes, to transfer 8 bit data requires 2 writes.

Now, the code you are pointing towards is using an LCM1602, but instead of direct GPIO connections, it's using an I2C expander.  This "pretends" to be an I2C device, but in actuality simply copies the 8-bit commands you send, to it's outputs (GPIOs) connected to the LCM1602 itself.  Since only 8 bits are available, the device MUST run in it's 4-bit mode.   So each 'command' is actually 4 bits of command GPIO values + 4 bits of data GPIO values.  You must still deal with the delays (sleeps) that the underlying LCM1602 module requires, whether you are doing direct GPIO control, or using a version with an I2C expander handling the front end interface.  Yes, this is hardly a speedy device.

Now, with the Grove RGB LCD (JHD1313M1) the situation is quite different.  It is essentially the same LCM1602 module, but with a much smarter I2C front end (ie: not just an 8 bit I2C expander).

If you look at the virtual command() and data() methods in the JHD1313M1 driver, you can see we do not need to deal with the 4-bit operating mode - we actually have 2 separate I2C registers, one for command functions and one for data - both 8bits wide.

So, perhaps some more detailed information would be useful.  I'm curious about "a bunch of reliability issues".  It's been awhile since I used one of these, but I don't recall there being any reliability issues reported, and it is a quite popular device.

jontrulson, 2016-11-03:
Hi, unless there is an actual issue here, I'd like to close this.  Any objections?

jontrulson, 2016-11-07:
Closing.  Please feel free to reopen if there is a problem we need to look at.

closed 2016-11-07 by jontrulson