eclipse-upm/upm #454: PN532 mraa_i2c_write() error with example script

matthieuglive, 2016-08-16:
Hi, I'm trying to use a PN532 RFID reader from deek-robot [http://www.aliexpress.com/store/product/MINI-PN532-NFC-RFID-Controller-Shield-for-Arduino/1178065_32213388605.html]  in i2c mode on a Galileo Gen 2 on yocto linux image with the included exemple that come with the library [https://github.com/intel-iot-devkit/upm/blob/master/examples/python/pn532.py] , but every time I get that error (See screenshot below) I posted on Intel Galileo forum about that, and Peter, a guy from intel answer me that he got the same issue and suggest me open an issue for the exemple script.

[image]

When i use te command i2cdetect -y -r 0 the RFID reader is reconized at the address 24 (whitch disapear when i unplug the reader) on the i2c-0 bus , So the bus seems to work or at least recognize the reader, but I'm unable to use it in the example script

 I use the pinout :

A4 -> SDA

A5 -> SCL

pin 2 ->RESET

pin 3 -> IRQ

MRAA and UPM are at their lastest versions  :

 
3.8.7-yocto-standard
MRAA v1.2.1
UPM v0.7.2
Node 4.4.3
 
I tested the RFID reader on an arduino UNO with an adafruit_PN532 library exemple and it works fine with the same pinout.
FYI, that arduino sketch DOESN'T works when I upload it to the arduino part the Galileo.

Propanu, 2016-08-23:
Hi and sorry for the wait.

>FYI, that arduino sketch DOESN'T works when I upload it to the arduino part the Galileo.

Does this not run for you? Were you able to compile it to begin with? There was another thread on the forums on this topic, see [here](https://communities.intel.com/thread/57654).

We developed this driver using the [Adafruit PN532 Breakout](https://www.adafruit.com/products/364) and the examples worked for both the Galileo G2 and Edison boards. Interestingly enough the shield version of the PN532 did give us problems on I2C writes too and I think the part you have was designed using Adafruit's shield as a reference. I suspect the issue might be caused by missing or poorly selected I2C pull-up resistors. While there is a chance this is software based, it will take some time to investigate.

Any additional details you can provide will help, thanks!

sahilkhanna, 2016-12-06:
Same thing happened to me(but on the edison) till I figure out what I was doing wrong.

After analysing the signals I found out that my I2C lines didnt have nay valid signals going through them. So I check my code again.
what I found was that I haven't set up my board connections properly. hence this is what I ended up using 

var myNFCObj = new pn532.PN532(MRAA_IRQ, MRAA_RESET, I2C_BUS_NUMBER);

closed 2018-09-30 by Propanu