eclipse-upm/upm #455: Docs for ULN200XA Stepper Driver

nevets963, 2016-08-18:
The ULN200XA Stepper Driver and Stepper Motor is included with the Grove Starter Kit and the docs I found are a bit unclear about how to wire it up to an Edison and also what 'VM' means (on the Stepper Driver): 

http://iotdk.intel.com/docs/master/upm/classupm_1_1_u_l_n200_x_a.html
https://software.intel.com/en-us/iot/hardware/sensors/uln200xa-stepper-driver

malikabhi05, 2016-08-18:
hi @nevets963 , regarding wiring up the Stepper Motor Driver, IN1 through IN4 are GPIO inputs which you can pass as arguments to the constructor ULN200XA::ULN200XA(int stepsPerRev, int i1, int i2, int i3, int i4). Vcc would have to be the 5V pin coming out of the Edison as for VM, it is an external 5V supply. You can try running the C++ or any other example provided for the motor in the UPM repo.
C++ example: https://github.com/intel-iot-devkit/upm/blob/master/examples/c%2B%2B/uln200xa.cxx

nevets963, 2016-08-22:
Hi @malikabhi05 Thank you for your quick reply. This all makes sense and I got it working - it just wasn't clear from the docs about the Vcc and VM. Maybe this could be added?

mrebb, 2017-01-31:
Can someone guide where exactly VCC, Gnd, VM has to be connected from stepper driver to edison?

Propanu, 2017-02-02:
@mrebb, connect either VCC or VM to the 5V pin and make sure the switch on the stepper motor driver matches the one you picked. GND goes to any GND pin on the Edison board.

malikabhi05, 2017-06-02:
Initial issue was resolved. Documentation added Vcc and Vm pins. 28f964cb48ae179c94103c9e70ed000ffad9554a

closed 2017-06-02 by malikabhi05