eclipse-upm/upm #456: About MPU9250

KaboChueng, 2016-08-19:
Hello!
  I want to use MPU9250 to get the posture of measured object. One solution is that I can get quaternion and Euler angle via dmp algorithm, but I didn't find dmp library.
  Does it exists? Or is there still another way to get quaternion and Euler angle?

Propanu, 2016-08-24:
Hi, check out @kriswiner's repos for some good examples on how to get sensor fusion with InvenSense IMUs. They are for Arduino, but our libraries should provide all the functionality needed to implement something similar.

And there's a great read on the subject in one of his [wiki](https://github.com/kriswiner/MPU-6050/wiki/Affordable-9-DoF-Sensor-Fusion) pages.

KaboChueng, 2016-08-25:
Thank you very much ! I have solved this problem.

KaboChueng, 2016-08-26:
Hi, I got a a new  problem. I can't use magnetometer,so I didn't get correct heading angle(yaw). I use Intel Edison board and Arduino expansion board . If the parameter of magnetometer in MPU9250's constructor is true, I get  an error like (I2C writeReg),so I have to set this parameter false.
Do you know some way to reduce deviation without magnetometer?

kriswiner, 2016-08-26:
If you want yaw (heading) that doesn't drift and is accurate to +/- 4
degrees or so after proper calibration, you really need the magnetometer.
Why not figure out why you can't read it?

heading angle(yaw). I use Intel Edison board and Arduio expansion board . If

nvL1ZBmks5qjoI5gaJpZM4Joe8U> .
<https://github.com/notifications/beacon/AGY1qkn53NNNqKkrPvbvKw06e9wGE0Lgks5
qjoI5gaJpZM4Joe8U.gif>

Propanu, 2016-08-29:
You might be able to use an I2C redriver/repeater/hub to resolve the I2C bus limitations on the Intel Edison with the Arduino breakout (e.g. [this](https://international.switch-science.com/catalog/2375)). However don't quote me on it as I haven't had the chance to try it yet.

KaboChueng, 2016-08-30:
Thank you! I will try it.

closed 2018-09-30 by Propanu