eclipse-upm/upm #457: HTU21D doesn't work with later mraa versions

arfoll, 2016-08-23:
Mraa's i2c error dection fixes have broken the HTU21D driver - see details here: https://communities.intel.com/thread/104939

whpenner, 2016-08-23:
I was making some changes on my weather system the other day and noted that the htu21d code has another problem.  About line 226/227 there are two - one byte register reads.  That should be one 16-bit read according to the documentation.  I was going to update the code, but my HTU21D part died and I haven't been able to get another up and running yet to verify the fixes.

Bill

Propanu, 2016-08-23:
We have a PR pending for this #433 but it isn't signed. @whpenner feel free to resubmit your version.

whpenner, 2016-08-24:
Ok, I'll probably send it up this evening or tomorrow morning.  I've made the code changes and need to finish testing it.

g-vidal, 2016-08-31:
Hi 
I am having great trouble with this sensor.... @arfoll pointed to my post in mraa discussions.
Would you please tell me if you can detect your HTU21D-F on the bus i2c with the standard i2cdetec command ?

I have checked various demo and startup programs from the arduino IDE, there are great differences including the one you are pointing out on the 2byte readings. There is another one concerning the release of the i2c bus which is necessary if another sensor is on the same bus. Have you seen this one (I found it on the datasheet)? Do you intend to amend it in your proposal ?

i have had no time to come into the code but I may find some time this weekend  if that helps, let me know. Thanks for your work.
Best regards
Gérard

g-vidal, 2016-08-31:
How can I remove my #433 proposal ? 
I am pretty sure that it solves only partially the problem  but I had not  been able to provide a complete solution....

Thanks

whpenner, 2016-08-31:
I have a pull request up that will make a change to the reset function.  It was missing the actual command to perform the reset.  I use this sensor in combination with other sensors without any issue.  You do need to make sure you have adequate pull-up resistors to either 3.3 or 5v (usually 3K - 10K) for both the SDA and SCL signals.

Bill

Propanu, 2016-08-31:
Hi @g-vidal, I've merged the fix from @whpenner (77fbf49f939e316d05942bd7752db29dd152ce60). Please let me know if the sensor is working for you as expected now. I will close #433 since the fix covers the changes in there too.

rista86, 2016-09-02:
When can we expect this package fixed on npm? Or can someone build the latest verison and provide us with the .node file so we can update manually? Thanx!

Propanu, 2016-09-02:
Hi @rista86. If you want to test this manually, download the latest development ipk from [here](http://iotdk.intel.com/repos/3.5/intelgalactic-dev/opkg/i586/upm_0.7.3.16_i586.ipk). It can be unzipped (twice) and you will find the .node module in there. Built with Node 4 for i586 (Galileo, Edison, Minnowboard 32-bit).

Meanwhile we are coming up with a generic solution for npm that will allow you to install the UPM modules regardless of Node version and architecture, but it will take a little longer. See #325 also.

g-vidal, 2016-09-04:
>
> Hi @g-vidal <https://github.com/g-vidal>, I've merged the fix from
> @whpenner <https://github.com/whpenner> (77fbf49
> <https://github.com/intel-iot-devkit/upm/commit/77fbf49f939e316d05942bd7752db29dd152ce60>).
> Please let me know if the sensor is working for you as expected now. I
> will close #433 <https://github.com/intel-iot-devkit/upm/pull/433>
> since the fix covers the changes in there too.
Yes @whpenner has really made a great work, congratulations and thanks!

I have cheked the code on a minibreakout with level shifter and with the
last mraa (compiled from git clone), it works perfectly ! @whpenner has
also included the hold mode which enables to have other sensors on the
i2c without any trouble. There is no need of pull-up resistor in my case.

It is still not possible to detect the HTU21D with the iodetect command
:-\ but every HTU21D user knows thet the adress is 0x40 then one can manage.
Thank you and thanks to @whpenner <https://github.com/whpenner>  you can
delete my first proposal.

best regards

Propanu, 2016-09-07:
I've seen at least 2 other sensors that don't show up in the table with i2cdetect but work fine when you start sending commands to them from MRAA. It might be very well the fact that i2cdetect tries to read a byte and some devices transfer a full word at a time, as @whpenner suggested on the forums.

Thanks a lot for confirming the sensor module works as expected after the patch!

closed 2016-09-07 by Propanu