eclipse-upm/upm #459: Building UPM on Ubuntu 14.04 failed

bso-intel, 2016-09-02:
Any idea how to fix this issue?

[ 34%] Built target _pyupm_micsv89
[ 34%] Building CXX object src/micsv89/CMakeFiles/micsv89.dir/micsv89.cxx.o
Linking CXX shared library libupm-micsv89.so
[ 34%] Built target micsv89
[ 34%] Swig source
/home/parallels/upm/upm-src/src/bno055/bno055.hpp:123: Error: Syntax error in input(3).
make[2]: *** [src/bno055/pyupm_bno055PYTHON_wrap.cxx] Error 1
make[1]: *** [src/bno055/CMakeFiles/_pyupm_bno055.dir/all] Error 2
make: *** [all] Error 2

malikabhi05, 2016-09-02:
Hi @bso-intel , what SWIG version are you using?

bso-intel, 2016-09-03:
Hi @malikabhi05,
I installed both swig2 and swig3 (3.0.2 => up-to-date).
On ubuntu, swig3.0.5 is not available, so I used cmake -DBUILDSWIGNODE=OFF option.

bso-intel, 2016-09-03:
Here is the output of verbose make command:
cd /home/parallels/upm/build/src/bmx055 && /usr/bin/swig2.0 -python -I/home/parallels/upm/build/src/bmx055/.. -outdir /home/parallels/upm/build/src/bmx055 -c++ -I/usr/local/include -I/home/parallels/upm/upm-src/src/bmx055/. -I/home/parallels/upm/upm-src/src/bmx055/.. -I/home/parallels/upm/upm-src/src/bmx055/.. -o /home/parallels/upm/build/src/bmx055/pyupm_bmx055PYTHON_wrap.cxx /home/parallels/upm/upm-src/src/bmx055/pyupm_bmx055.i
/home/parallels/upm/upm-src/src/bmx055/bma250e.hpp:84: Error: Syntax error in input(3).
make[2]: *** [src/bmx055/pyupm_bmx055PYTHON_wrap.cxx] Error 1
make[2]: Leaving directory `/home/parallels/upm/build'
make[1]: *** [src/bmx055/CMakeFiles/_pyupm_bmx055.dir/all] Error 2
make[1]: Leaving directory `/home/parallels/upm/build'
make: *** [all] Error 2
pa

Propanu, 2016-09-14:
You will have to disable Python bindings too with `-DBUILDSWIGPYTHON=OFF`.
SWIG >= 3.0.5 is required for the bindings, and is available on some PPA's. From our Travis CI config:
```
sudo add-apt-repository --yes ppa:fenics-packages/fenics-dev/swig
sudo apt-get install -y --force-yes -qq swig3.0 git
```
The latest version can also be built from [source](http://www.swig.org/download.html) relatively painless.

pylbert, 2016-10-12:
@bso-intel, closing issue.  If you would like to build the python/java/nodejs modules let me know and I can help.

closed 2016-10-12 by pylbert