eclipse-upm/upm #460: Cannot writeCalibrationData() for BNO-055

tripish, 2016-09-07:
After the calibration sequence is completed for the sensor, executing the following command
```
sensor.writeCalibrationData(sensor.readCalibrationData())
```

Returns the error
```
Error: UPM Invalid Argument: writeCalibrationData: calibData string must be exactly 22 bytes long
    at Error (native)
    at repl:1:8
    at REPLServer.defaultEval (repl.js:262:27)
    at bound (domain.js:287:14)
    at REPLServer.runBound [as eval] (domain.js:300:12)
    at REPLServer.<anonymous> (repl.js:431:12)
    at emitOne (events.js:82:20)
    at REPLServer.emit (events.js:169:7)
    at REPLServer.Interface._onLine (readline.js:211:10)
    at REPLServer.Interface._line (readline.js:550:8)
```

My goal is to record the calibration data to a file on my device, so that I can upload the calibration data upon startup.  The issue it is seems there's something wrong either with the writeCalibrationData() or readCalibrationData() functions.

I haven't built a UPM module myself yet to try printing logs and playing around, but I'm wondering if the issue is in bno055.cxx, either ln 471 where readCalibrationData() casts the uint8_t array to a string, or ln 484 where writeCalibrationData() checks the size of the incoming calibration string.

Anyone see something obviously wrong here?

Propanu, 2016-09-21:
Perhaps issue with string not being null terminated after cast? @jontrulson any ideas?

jontrulson, 2016-09-21:
Hi, the string is just being used to transport binary data, so there is no 0 termination (and should not be any).  Looking at the code it is not at all clear what could be going on.  The next step is to look at writeCalibrationData() and see what the passed string size() actually is.

tripish, 2016-11-16:
I was playing around and seem to have the following running correctly:
```
var calibrationData = Buffer.from(sensor.readCalibrationData(),'ascii');
sensor.writeCalibrationData(calibrationData.toString('ascii'));
```

My problem was the default Buffer creation from a string used utf-8 encoding, and turned the calibration string into 42 bytes instead of 22.

closed 2016-11-16 by tripish