eclipse-upm/upm #461: Doing add-ons of the UPM sensor (smartdrive) to javascript and python

neuberfran, 2016-09-08:
Hi guys,

I did this driver ( https://github.com/intel-iot-devkit/upm/blob/master/examples/c%2B%2B/smartdrive.cxx) in partnership with driver
Oussema Harbi <oussema.elharbi@gmail.com>

What I need now to do  **_var smart= require(‘smartdrive’)_** and use smart variable to move windshield motors to left and right in my javascript code?

e.g.:  SmartDrive High Current Motor Controller  is originally sold with a python software driver to raspberry pi. I worked in last year to  solve problems and make the driver works in intel edison
(https://communities.intel.com/thread/88007)

pylbert, 2016-09-14:
That *almost* correct.  

var lib = require('jsupm_smartdrive');
var smartdrive = new lib.SmartDrive(0);

It might be nice to have the corresponding java/javascript/python examples as well.

neuberfran, 2016-09-20:
https://drive.google.com/file/d/0B9gPdWq4OH-XakU5RTJ4N2xjbWc/view?usp=sharing

Now I stay trying install smartdrive (c++ version) in new edison with new yocto image (201606061707) and I'm having problems:
CMake Error at CMakeLists.txt:6 (upm_module_init):
  Unknown CMake command "upm_module_init"

Propanu, 2016-09-21:
You need the full UPM source tree to start a build with CMake. If you just need the shared objects I recommend using opkg as described in [this article](https://software.intel.com/en-us/node/623543). To update MRAA use `mraa` instead of `libmraa0` though.

neuberfran, 2016-09-21:
Ok, but no I have new issue

https://drive.google.com/file/d/0B9gPdWq4OH-XQ3NrM2ZyR3Z4V3M/view?usp=sharing

neuberfran, 2016-09-23:
https://communities.intel.com/thread/106315

neuberfran, 2016-09-23:
https://communities.intel.com/inbox

neuberfran, 2016-09-23:
File smartdrive.cxx this in intel edison with oldest image and running OK
 
http://pastebin.com/4RFjFPKd
When I pu this file in new intel Edison  I got these errors
 
**root@neuberfran:~/upm/examples/c++#** g++ smartdrive.cxx -o testSmartDrive -lupm-smartdrive -I/usr/local/include/upm;
/tmp/ccziaZ1l.o: In function `main':
smartdrive.cxx:(.text+0xcc): undefined reference to `upm::SmartDrive::StopMotor(MotorID_t, MotorAction_t)'
smartdrive.cxx:(.text+0x136): undefined reference to `upm::SmartDrive::ReadTachometerPosition(MotorID_t)'
smartdrive.cxx:(.text+0x17e): undefined reference to `upm::SmartDrive::ReadTachometerPosition(MotorID_t)'
smartdrive.cxx:(.text+0x1d0): undefined reference to `upm::SmartDrive::Run_Seconds(MotorID_t, Direction_t, unsigned char, unsigned int, bool, MotorAction_t)'
smartdrive.cxx:(.text+0x22d): undefined reference to `upm::SmartDrive::ReadTachometerPosition(MotorID_t)'
collect2: error: ld returned 1 exit status
 
2) File smartdrive.cxx this in intel Edison with new image but don’t running
 
http://pastebin.com/bV0tm3xT
When I pu this file in new intel Edison  I got these errors
 
**root@neuberfran:~/upm/examples/c++#** g++ smartdrive.cxx -o testSmartDrive -lupm-smartdrive -I/usr/local/include/upm;
/usr/lib/gcc/i586-poky-linux/4.9.1/../../../../i586-poky-linux/bin/ld: /tmp/ccoq6w3k.o: undefined reference to symbol 'mraa_i2c_stop'
//usr/local/lib/libmraa.so.1: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
root@neuberfran:~/upm/examples/c++# g++ smartdrive.cxx -o testSmartDrive -lupm-smartdrive -I/usr/local/include/upm;
/usr/lib/gcc/i586-poky-linux/4.9.1/../../../../i586-poky-linux/bin/ld: /tmp/cc8gE5Cn.o: undefined reference to symbol 'mraa_i2c_stop'
//usr/local/lib/libmraa.so.1: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status

neuberfran, 2016-09-27:
Please change this code (wrong code (1))  in github: http://pastebin.com/bV0tm3xT

to
this:

http://pastebin.com/4RFjFPKd
(this code worked in old intel edison with old image)

(1)
[image]

Propanu, 2016-09-29:
@neuberfran, the goal here would be to get the example working with the latest image and libraries. Were you able to build the example after all? The linker errors you show indicate you should add the `-lmraa` flag as well.

neuberfran, 2016-09-30:
https://communities.intel.com/thread/106315?start=15&tstart=0

Thank's. I put -lmraa flat and I have new erros with code in github 
[image]

neuberfran, 2016-10-02:
I reinstalled the image, and now have old issues
(motors don't move)
backupasteriskATELATALnewERROerro2.png

neuberfran, 2016-10-02:
[image]

neuberfran, 2016-10-02:
When motors is running (in old edison with old image) the image is:
[image]

neuberfran, 2016-10-04:
https://communities.intel.com/thread/106315?start=15&tstart=0

neuberfran, 2016-10-04:
**_When I put this code I have this errors:_** 

var smart = require('jsupm_smartdrive');
var smarttest = new smart.SmartDrive(0);
smarttest.SmartDrive();

[image]

neuberfran, 2016-10-06:
Hi,

I get voltage But I can't run motors with this code:
  
var smart = require('jsupm_smartdrive');
var smarttest = new smart.SmartDrive(0);
var teste = smarttest.GetBattVoltage();
smarttest.Run_Seconds(0x01, 0x01, 90, 70, true, 0x01);
console.log(teste);

[image]

neuberfran, 2016-10-08:
Now motors run When I put SmartDrive_Motor_ID_BOTH But not run When I put SmartDrive_Motor_ID_1 our 2:

[image]

neuberfran, 2016-10-10:
The smartdrive.cxx of the /home/root/upm/src/smartdrive/ of the old edison with old image (I am calling smartdrive.cxx principal) : http://pastebin.com/R0E0jpCm (special attention to this part:

run motors 1 and/or 2 separately
 
I think that this code has to be put in place this on github: upm/smartdrive.cxx at master · intel-iot-devkit/upm · GitHub

Propanu, 2016-10-14:
@neuberfran thanks for following up. I'll check on your suspicion that `writeArray()` only writes one byte due to returning the pointer size instead. Can the shield run the motors even with default values set for the control registers? That would explain why both start on `writeByte(SmartDrive_COMMAND, CMD_S);`. Unfortunately, I don't have one of these shield to see for myself.

neuberfran, 2016-10-14:
The answer to your question is not. Can't the shield run the motors even with default values set for the control registers.

errors When I can change smartdrive.cxx of the upm/src/smartdrive

[image]

The tests to which I referred in my previous post I made this in old intel edison with old image and old /upm/src/smartdrve/smartdrive.cxx

neuberfran, 2016-10-14:
var smart = require('jsupm_smartdrive');
var smarttest = smart.SmartDrive;
var smarttest = new smart.SmartDrive(0);
var teste = smarttest.GetBattVoltage();
 
**_This code above/below not run motor 02 (in intel edison with new image) even after riding the ./testSmartDrive witho _BOTH option_**
 
var smart = require('jsupm_smartdrive');
var smarttest = smart.SmartDrive;
var smarttest = new smart.SmartDrive(0);
var teste = smarttest.GetBattVoltage();
 
setTimeout(function(){
    smarttest.Run_Seconds(0x02,0x00, 90, 70, true, 0x01 );
 
 
    console.log(teste);
    setInterval(function(){
           smarttest.StopMotor(0x02,0x02);
    }, 4000);
}, 4000);

neuberfran, 2016-10-18:
I change write array function to use static size (5),and yet don't move motor 01 or 02.
 
When buying the smartdrive sensor from openelectronics, the original APi is python.
 
I'm use this code: https://github.com/neuberfran/SmartDrive-demo/blob/master/SmartDrive-demo.py

in edison old with old image (where this functional, because  not just take a pip install smartdrive in any edison (instead pi raspberry)  had to move in smbus
and such code worked (move motor 01)
Therefore I need your help to change this code: https://github.com/intel-iot-devkit/upm/blob/master/src/smartdrive/smartdrive.cxx
based on that: https://github.com/neuberfran/SmartDrive-demo/blob/master/SmartDrive.py

mainly in two parts:

[image]

and when other programmer use this: if ( motor_id != SmartDrive_Motor_ID_2) {
or this if ( motor_id != SmartDrive_Motor_ID_1){
because in Smartdrive.py Never see it

Propanu, 2016-10-26:
Hi @neuberfran, I can confirm that the `writeArray()` function is wrong. I will update the code for it to take in the length of the array as well. The rest of the logic seems OK, matching the Python implementation, but I won't be able to test without the actual shield which is unfortunately out of stock. Your continued help on this is greatly appreciated.

neuberfran, 2016-10-27:
http://www.mindsensors.com/rpi/76-smartdrive-high-current-motor-controller

neuberfran, 2016-11-01:
Hi,

They made the code update, included **_extended writeArray() function to use proper array length_**
and I do this again: 
[image]

**_with this code:_**
https://github.com/neuberfran/smartdriveNEVERrun/blob/master/smartdrive.cxx

But motor_02 or motor_01 never Run

Propanu, 2016-11-03:
Hi @neuberfran, sorry to hear the `writeArray()` fix didn't work. That was definitely wrong, but apparently there's more to it. Since the shield is still out of stock, the next step I can take is to compare the I2C commands sent by the driver in our library vs. the original Python driver. Will follow up once I have something.

neuberfran, 2016-11-03:
Thank's, but
you are right. **Code needs more corrections.**
This code with Python api of the openelectrons that I could make it run in edison with old image **is now working perfectly**

https://github.com/neuberfran/SmartDrive-issue/blob/master/SmartDrive-demo.py

https://communities.intel.com/thread/106315?start=45&tstart=0

[image]

Links of the Python APi of the openelectrons/mindsensor

https://github.com/neuberfran/SmartDrive-demo/blob/master/SmartDrive.py
https://github.com/neuberfran/SmartDrive-issue/blob/master/OpenElectrons_i2c.py
https://github.com/neuberfran/SmartDrive-demo/blob/master/smbus.py
https://github.com/neuberfran/SmartDrive-issue/blob/master/__init__.py
https://github.com/neuberfran/SmartDrive-issue/blob/master/util.py

neuberfran, 2016-11-08:
Hi,

Please change https://github.com/intel-iot-devkit/upm/blob/master/src/smartdrive/smartdrive.cxx and https://github.com/intel-iot-devkit/upm/blob/master/src/smartdrive/smartdrive.hpp  **for** **this** 

https://github.com/neuberfran/SmartDriveCorrect/blob/master/smartdrive.cxx  and this https://github.com/neuberfran/SmartDriveCorrect/blob/master/smartdrive.hpp

**_I did several tests and I advise that the best thing to do is_**

neuberfran, 2016-12-27:
Pls

How to I use mraa/upm (Smartdrive) in intel edison with Android Things SO

neuberfran, 2017-05-13:
Oi,

Altere https://github.com/intel-iot-devkit/upm/blob/master/src/smartdrive/smartdrive.cxx e https://github.com/intel-iot-devkit/upm/blob/master/ Src / smartdrive / smartdrive.hpp para isso

Https://github.com/neuberfran/SmartDriveCorrect/blob/master/smartdrive.cxx e este https://github.com/neuberfran/SmartDriveCorrect/blob/master/smartdrive.hpp

Eu fiz vários testes e aconselho que a melhor coisa a fazer é

closed 2018-09-30 by Propanu