eclipse-upm/upm #470: Symbol lookup error for GroveTemp in C++

deadprogram, 2016-10-20:
While attempting to run the https://github.com/intel-iot-devkit/how-to-code-samples/tree/master/fire-alarm/cpp sample, I am receiving the following error:

```
/tmp/Fire_Alarm: symbol lookup error: /tmp/Fire_Alarm: undefined symbol: _ZN3upm9GroveTempC1Ejf
```

This is the same problem as reported in https://communities.intel.com/thread/105797

Seems like perhaps the `GroveTemp` symbol was not exported into the `upm-grove` lib as it previously had been?

Propanu, 2016-10-20:
It's still there but constructor signature was changed when I pushed bda66a3d06ee205f6f24fe2441cecece4ffed2fb (`00003bf0 g    DF .text  00000180  Base        _ZN3upm9GroveTempC1Ejfii`). Binary compatibility was broken but sources will still compile without any change required. I'll add to API changes list, thanks for pointing it out!

deadprogram, 2016-10-31:
Is this going to be fixed? I see that the new version of UPM was just released, but seems like the function signature still needs update, or ?

Thank you!

Propanu, 2016-11-02:
If this is still an issue for you, the libraries on your board are most likely out of sync with your host. When using ISS to compile projects, make sure it updates the libraries from [IoTDK 3.5](http://iotdk.intel.com/repos/3.5/intelgalactic/opkg/i586/) and not the 3.0 version of the repos. The older version doesn't go past UPM 0.7.2 (hence why the old symbol). Can be changed in IDE settings.

deadprogram, 2016-11-02:
Where in the IDE settings for ISS do I change that setting? I looked under "Connection > Open Package Preferences..." and "Intel IoT > Managed Installed Development Platforms..." and "Intel IoT > Packages Update..."

Note that when I "Synchronize Packages" using the context menu on the connection to the gateway, I see this output in the terminal from the gateway:

root@WR-IDP-FF4F:~# chmod 777 /tmp/smart-sync-detection.sh; /tmp/smart-sync-dete ction.sh mraa upm ; exit;

1.3.0:0.8.0

The dialog box displayed by ISS shows the docker container has mraa 1.0.01.3.0 and upm 0.7.00.8.0. The target says "Can't detect target version".

deadprogram, 2016-11-03:
@tingleby were you going to chime in on this? :angel:

tingleby, 2016-11-03:
@deadprogram  ..Yes..

The docker container would be out of date. A patch is online on docker hub now. Please update the Gateway 64 docker image.
mraa 1.3.0
upm 0.8

deadprogram, 2016-11-03:
How should I update the docker image?

deadprogram, 2016-11-03:
@tingleby can I just do a `docker pull inteliotdevkit/intel-iot-wrs-64:latest` to update the image?

tingleby, 2016-11-03:
Yes

tingleby, 2016-11-03:
Also you may have to create a new work space OR toggle in out of the Gateway container.

deadprogram, 2016-11-04:
This fixed the problem. Thank you very much.

tingleby, 2016-11-04:
No problem

closed 2017-01-10 by deadprogram