eclipse-upm/upm #475: python upm module location/names

arfoll, 2016-10-26:
Python modules have been moved to a upm/ subdir which is pretty nice actually, but why has the pyupm_ prefix been retained? That seems complete overkill and would make it much nicer to do:

```
from upm import i2clcd as lcd
```

instead of 
```
from upm import pyupm_i2clcd as lcd
```

g-vidal, 2016-10-26:
Hi all,
I second this proposal :-)

jontrulson, 2016-10-26:
This naming pre-dates my involvement with this project, but I think doing this where possible would be a nice improvement.  Of course it would be an "API" change though right?  With all that might entail...

pylbert, 2016-10-26:
Correct.  This would affect all the SWIG module usage.  The xxxupm_ strings are everywhere in the UPM repo:
CMakeLists.txt
SWIG .i files
Examples
etc...

It would be nice to see more uniform usage throughout.  Cmake targets could also benefit from a renaming to simplify the build environment.

Example target names for dfrph (current):
_pyupm_dfrph-python3
_pyupm_dfrph-python2
javaupm_dfrph
dfrph
dfrph-c
jsupm_dfrph

Could be:
dfrph-c
dfrph-cxx
dfrph-java
dfrph-node
dfrph-python2
dfrph-python3

jontrulson, 2016-10-26:
> dfrph-c
>dfrph-cxx
>dfrph-java
>dfrph-node
>dfrph-python2
>dfrph-python3

Yes, I like that.  
We could also rename the driver interface files - 

jsupm_mma7660.i -> javascript.i
javaupm_mma7660.i -> java.i
pyupm_mma7660 -> python.i

In the driver dirs too.

Propanu, 2016-10-26:
I think these are all good changes and I agree with them. However, time constraints will prevent us from being able to integrate and test them for the upcoming 1.0 tag. We can go down this route after our next stable release.