eclipse-upm/upm #480: Some Grove-independent JS examples still using jsupm_grove

SuyashLakhotia, 2016-11-03:
I see that there are separate JS snippets for Grove sensors and otherwise. However, looks like some of the non-Grove snippets also seem to be using `jsupm_grove` instead of `jsupm_<sensor>`, which means that when the sensors are being initialised (as JS objects), an incorrect constructor is being called. I've gone through some of the files really quickly and found that this mistake exists in:

- [Grove Button](https://github.com/intel-iot-devkit/upm/blob/master/examples/javascript/grovebutton.js) vs. [Button](https://github.com/intel-iot-devkit/upm/blob/master/examples/javascript/button.js)
- [Grove Rotary](https://github.com/intel-iot-devkit/upm/blob/master/examples/javascript/groverotary.js) vs. [Rotary](https://github.com/intel-iot-devkit/upm/blob/master/examples/javascript/rotary.js)

I'll build the above list if I find any more. I can also submit a PR fixing the issue if you wish.

On a side note, some of the correct usages can be found in:

- [Grove LED](https://github.com/intel-iot-devkit/upm/blob/master/examples/javascript/groveled.js) vs. [LED](https://github.com/intel-iot-devkit/upm/blob/master/examples/javascript/led.js)
- [Grove Temp](https://github.com/intel-iot-devkit/upm/blob/master/examples/javascript/grovetemp.js) vs. [Temp](https://github.com/intel-iot-devkit/upm/blob/master/examples/javascript/temperature.js)

Closes #474.

jontrulson, 2016-11-03:
@SuyashLakhotia , thanks for locating these, I guess a couple slipped through during the split.  If you'd like to submit a PR fixing them, please feel free to do so.  :)

Thanks!

SuyashLakhotia, 2016-11-03:
@jontrulson Done. Fixed in #481. :)

Propanu, 2016-11-03:
@SuyashLakhotia, some of the documentation and the samples out there will be updated over time, but I highly recommend moving to the new class names. As for how we group those into packages, we are working on that and I agree with you that it makes sense to provide kit specific library bundles.

closed 2016-11-08 by SuyashLakhotia