eclipse-upm/upm #482: ads1115 support broken in js

netvandal, 2016-11-03:
Hi, Just tried out the sample 
https://github.com/intel-iot-devkit/upm/blob/master/examples/javascript/ads1x15.js
and it is not working with libmraa 1.4.0 and libupm 1.0.0 on Intel Edison. 

Program die with:

var ads1115 = new ADS1X15.ADS1115(1, 0x49); 
              ^

Error: Illegal arguments for construction of _exports_ADS1115
    at Error (native)
    at Object.<anonymous> (/home/root/.node_app_slot/main.js:17:15)
    at Module._compile (module.js:409:26)
    at Object.Module._extensions..js (module.js:416:10)
    at Module.load (module.js:343:32)
    at Function.Module._load (module.js:300:12)
    at Function.Module.runMain (module.js:441:10)
    at startup (node.js:139:18)
    at node.js:968:3

Propanu, 2016-11-03:
Hi, is this on the Intel Edison with the Arduino breakout? If so, you'll want to use i2c bus 0 instead, i2c-1 is not exposed on the headers.

Propanu, 2016-12-15:
@netvandal, were you able to resolve this? As a note, you might also have to change the I2C address if it's different for your ads1115 (it's configurable on these ADCs).

netvandal, 2017-02-10:
@Propanu yes, just need to use 0 in the bus parameter, while the address was right. 
var ads1115 = new ADS1X15.ADS1115(0, 0x48);  

This using this AdaFruit board: https://www.adafruit.com/product/1085 as reference
Sorry for the delay in the response.

closed 2017-02-10 by netvandal