eclipse-upm/upm #484: Compilation errors in upm lib "error: this ‘if’ clause does not guard..." (bmi160,speaker,...)

g-vidal, 2016-11-06:
Hi,
I am used to compile upm last version from git but this time I get repeated errors from various sources. As I do not need them i remove the source and restart compilation but there are more and more occuring : 

- bmi160
- speaker
- smartdrive

Here is the error that look likes an indentation problem and sorry for the comments in french

`upm/src/smartdrive/smartdrive.cxx: In member function ‘void upm::SmartDrive::Run_Tacho(int, uint8_t, uint32_t, bool, int)’:
upm/src/smartdrive/smartdrive.cxx:330:9: error: this ‘if’ clause does not guard... [-Werror=misleading-indentation]
         if ( wait_for_completion )
         ^~
upm/src/smartdrive/smartdrive.cxx:332:13: note: ...this statement, but the latter is misleadingly indented as if it is guarded by the ‘if’
             WaitUntilTachoDone(motor_id);
             ^~~~~~~~~~~~~~~~~~
[ 20%] Linking CXX shared library libupm-sm130.so
Scanning dependencies of target st7735
cc1plus: all warnings being treated as errors
[ 20%] Building CXX object src/st7735/CMakeFiles/st7735.dir/st7735_gfx.cxx.o
src/smartdrive/CMakeFiles/smartdrive.dir/build.make:62 : la recette pour la cible « src/smartdrive/CMakeFiles/smartdrive.dir/smartdrive.cxx.o » a échouée
make[2]: *** [src/smartdrive/CMakeFiles/smartdrive.dir/smartdrive.cxx.o] Erreur 1
CMakeFiles/Makefile2:28405 : la recette pour la cible « src/smartdrive/CMakeFiles/smartdrive.dir/all » a échouée
make[1]: *** [src/smartdrive/CMakeFiles/smartdrive.dir/all] Erreur 2
make[1]: *** Attente des tâches non terminées....
[ 20%] Building CXX object src/st7735/CMakeFiles/st7735.dir/st7735.cxx.o
[ 20%] Built target sm130
[ 20%] Linking CXX shared library libupm-ssd1351.so
[ 20%] Built target ssd1351
[ 20%] Linking CXX shared library libupm-st7735.so
[ 20%] Built target st7735
Makefile:127 : la recette pour la cible « all » a échouée
make: *** [all] Erreur 2
`
Can I do anything to prevent this error message to happen and stop compilation  or should I try to amend the source code and commit?
Thanks again for the great job done
Best regards

pylbert, 2016-11-06:
@g-vidal, Werror has been turned on by default for UPM.  This can be disabled via -DWERROR=off.

Gcc 6 has -Wmisleading-indentation included in -Wall.  I will merge a fix to turn off the misleading indentation warnings.

Thanks,
noel

g-vidal, 2016-11-07:
Many thanks @pylbert for quick answer and workaround.
Should I wait you confirmed your merge to close the issue or should I/you close it now.
G.

pylbert, 2016-11-07:
Sorry, I haven't made the change yet. We can wait to close the issues till
the fix is on place.

pylbert, 2016-11-08:
@g-vidal, fixed in 68091dcf4317043a09d7684cb467fe0cb4fa738d

closed 2016-12-08 by Propanu