eclipse-upm/upm #486: UPM Issues Following Upgrade to MRAA 1.5.1 and UPM 1.0.0

dhrynyk, 2016-11-14:
On the Intel Edison I upgraded to MRAA 1.5.1 and UPM 1.0.0.  Following the upgrade calls to the C++ UPM device class Lcm1602 failed with a Segmentation fault.  Prior to the upgrade the application performed without error. 

Running the application under GDB I discovered the fault took place on the following call:

120	    lcd = new upm::Lcm1602(6, 0x27, true, 20, 4);

The diagnostic message was: 

Program received signal SIGSEGV, Segmentation fault.
0xb7fa18d7 in mraa_i2c_address () from /usr/lib/libmraa.so.1
 
Running the application under Valgrind yielded further diagnostic information of the fault:

==5640== Invalid write of size 4
==5640==    at 0x433DEB3A: std::basic_string<char, std::char_traits<char>, std::allocator<char> >::basic_string() (in /usr/lib/libstdc++.so.6.0.20)
==5640==    by 0x407A28C: upm::LCD::LCD() (in /usr/lib/libupm-i2clcd.so.1.0.0)
==5640==    by 0x407A4BB: upm::Lcm1602::Lcm1602(int, int, bool, unsigned char, unsigned char) (in /usr/lib/libupm-i2clcd.so.1.0.0)
==5640==    by 0x804964F: main (distance.cpp:120)
==5640==  Address 0x4282e1c is 0 bytes after a block of size 4 alloc'd
==5640==    at 0x40272EC: operator new(unsigned int) (in /usr/lib/valgrind/vgpreload_memcheck-x86-linux.so)
==5640==    by 0x8049637: main (distance.cpp:120)
==5640== 
==5640== Invalid read of size 4
==5640==    at 0x433E05BD: std::string::assign(char const*, unsigned int) (in /usr/lib/libstdc++.so.6.0.20)
==5640==    by 0x433E0798: std::string::operator=(char const*) (in /usr/lib/libstdc++.so.6.0.20)
==5640==    by 0x407A2A5: upm::LCD::LCD() (in /usr/lib/libupm-i2clcd.so.1.0.0)
==5640==    by 0x407A4BB: upm::Lcm1602::Lcm1602(int, int, bool, unsigned char, unsigned char) (in /usr/lib/libupm-i2clcd.so.1.0.0)
==5640==    by 0x804964F: main (distance.cpp:120)
==5640==  Address 0x4282e1c is 0 bytes after a block of size 4 alloc'd
==5640==    at 0x40272EC: operator new(unsigned int) (in /usr/lib/valgrind/vgpreload_memcheck-x86-linux.so)
==5640==    by 0x8049637: main (distance.cpp:120)
==5640== 

Plus many more messages.

Additional Information which may be useful:
•	Upgraded from MRAA 1.4.0 and UPM 0.8.0
•	Current Firmware Release: 3.5
root@edison:~# cat /etc/version
201606061707	

System Journal (journalctl) was not available:

root@edison:~# journalctl -f
No journal files were found.

I have reverted to MRAA 1.4.0 and UPM 0.8.0 and the application is now functioning.

Any assistance would be appreciated.

jontrulson, 2016-11-16:
Hi, this seems to be a problem with a corrupted malloc arena taking place before the constructor is called.  It would be nice to see a gdb stack trace, and your actual code.  

From what you've provided, it looks like the damage is already done by the time the LCM1602() ctor is called.

Assigning a string, such as `std::string m_name = "Hi there"` is not a complicated operation, but does require an allocation.  This particular assignment in the ctor has been there for years, and cannot be the cause of this problem.

mraa_i2c_address() does not do any allocation, but does operate on previously allocated memory (m_i2c_lcd_control), which appears to be corrupted by the time it is called.

Please try the lcm1602-i2c.cxx standalone example and see if that works for you.  My bet is that there is something in your code that was just accidentally working in previous versions.

dhrynyk, 2016-11-17:
Hi Jon:

I see your point and agree that the issue may reside in the code executed prior to the call to lcm1602. I can think of a number of tests I can run to track down the issue.  I need to wrap up some simple work on mraa 1.4.0 and then will upgrade and test.

In the meantime I would appreciate your thoughts on address verification.  Currently I'm using VALGRIND to validate for memory addressing issues. I've also read about address sanitizer (ASAN).  Any thoughts on an appropriate tool to track down addressing related issues?

Thank you,
David

dhrynyk, 2016-11-23:
Hi Jon:

I have been able to recreate the error using the standalone example.  In this case I was using MRAA 1.5.1 and UPM 1.0.1.  In the attached file I have provided:
1. Backgrace and Memory Map
2. First set of VALGRIND diagnostics 
3. Application listing
4. System Log Contents (journalctl -f) at the run time.

The example ran successfully using MRAA 1.4.0 and UPM 0.8.0.  

I wonder if there are any library dependencies with UPM 1.0.x.

Any insights or suggestions would be welcomed. 

Thank you,
David
[Lcm1602 UPM 1.0.1 Error.docx](https://github.com/intel-iot-devkit/upm/files/607887/Lcm1602.UPM.1.0.1.Error.docx)

dhrynyk, 2016-11-23:
Hi Jon:

I believe I now understand the issue.  In UPM version 0.8.0 I used the library 'upm-i2clcd' and that worked.  In version 1.0.x the library changed to 'upm-lcm1602'. I have made the change and the standalone example now works.

Please let me know if I've missed anything  else.

Thanks
David

jontrulson, 2016-11-28:
Hi, sorry for the delay in response, I was on vacation.  I'm glad you figured it out.  We kept the older driver in the upm-i2clcd library to try to preserve backward compatibility for a while but the interfaces (.hpp) are not quite the same.  We may need to consider removing it from the older library.

dhrynyk, 2016-11-30:
Hi Jon:

I do appreciate you getting back to me and providing the background.  I agree removing the older driver from the library may be the right approach.

Thanks
David

closed 2016-11-30 by dhrynyk