eclipse-upm/upm #487: Wrong sign for gravity on MPU9250

thoyo, 2016-11-15:
I am getting positive values for the acceleration's Z component when placing MPU9250 on a flat surface (and following the criteria from the datasheet here, section 9.1: http://wiki.seeedstudio.com/images/5/53/MPU-9250A_Product_Specification.pdf). 

According to the datashet, gravity should be negative.  

To retrieve the data I am using the Python code from here:
https://software.intel.com/en-us/iot/hardware/sensors/mpu9250-inertial-measurement-unit

 I wonder if there might be an issue on the data retrieve.

jontrulson, 2016-11-16:
Hi, you might be misunderstanding the diagram.

Picture taking the device into space, with 0G.  Then, with the device facing up as in the diagram, push the device "upwards" in the +Z direction, and you will get a +Z acceleration value.

With gravity in the mix, you can picture gravity as a continuous acceleration, in this case, the positive Z direction.  Both the datasheet, and the code are correct.

Einstein proved this relationship between gravity and acceleration to be true :)  Rather than gravity "pulling you down", view it as if you are constantly accelerating in the +Z direction to remain in position.

thoyo, 2016-11-17:
That makes a lot of sense. Thanks Jon!

closed 2016-11-17 by thoyo