eclipse-upm/upm #489: ds18b20 - Undefined Symbol error

sanjum89, 2016-11-18:
DS18b20 sensor was working using UART and this library. But currently it throws an error at the require line.
var ds= require('jsupm_ds18b20')

I've attached screenshot of the error. Any help is appreciated.

Thanks,

[image]

benjaminmathews, 2016-12-06:
I'm working with OP on this project. We had opened up a discussion for this issue in the Intel community here: https://communities.intel.com/thread/108426
Managed to find the issue and a solution for our particular use case. It wasn't an issue with the library, it seems. Hope that helps!

Propanu, 2016-12-15:
Hi @sanjum89 and @benjaminmathews, glad to hear this was resolved. As I mentioned in the other thread, #497, the undefined symbols would still cause issues if the MRAA/UPM shared libraries are not part of the image you are using. The sources on NPM have been updated to 1.0.2-src, and they include a fix for this.

closed 2017-01-23 by Propanu