eclipse-upm/upm #494: ds18b20 object creation hangs

ericksonla, 2016-11-29:
I'm trying to get a ds18b20 temperature sensor up and running using the [example python code from this repository](https://github.com/intel-iot-devkit/upm/blob/master/examples/python/ds18b20.py). The line `sensor = sensorObj.DS18B20(0)` is hanging. 

It looks like this should be initializing the uart connection. Does that suggest I have a problem with the physical connection? I'm a little confused by the 0 terminology because the [Edison breakout board hardware guide](http://download.intel.com/support/edison/sb/edisonbreakout_hg_331190006.pdf) only specifies a uart1 and uart2, not 0.

Any suggestions on what's going on here?

Propanu, 2016-12-03:
Hi @ericksonla, in theory it works on any UART, just make sure you match the bus number with the constructor and that you have the external circuit in place. Also, it might help to take a look at #421 to see if you are not in one of the cases described there. The connection diagram is at the very bottom and also on our [known limitations](../blob/master/docs/knownlimitations.md) list.

ericksonla, 2016-12-22:
Turns out I was using an incompatible level shifter. I switched to the TXS010x and now its working.

It would be nice if the initialization process had a timeout that reported some sort of debug information for future problems though. Any chance of getting that?

Propanu, 2017-01-23:
After a quick code check I couldn't find any places where the UPM driver or MRAA could get stuck unless this was something @jontrulson addressed in his last patch to the OW driver. Perhaps this happens at a lower level on some of the system calls, putting it beyond our control. Anyway, glad you figured it out. @jontrulson any comments on this before I close it?

jontrulson, 2017-01-23:
The only place I think this could happen would be in MRAA, _ow_read_byte(), which spins waiting for a byte (any byte).  If a situation on the bus prevented a character from ever being read, there might be an issue there.  Duplicating it so that we could find out where it hangs would be the way to go, then we might be able to add some sort of time out or something.

But if the circuit is wired correctly, I have never seen such a hang.  Of course, I have not deliberately tried to mess up the circuit, which might be why I've not seen this.

Propanu, 2017-01-23:
Yes, it only hangs if the circuit is not properly realized. If this can be easily fixed with a counter rather than timers or such I would say let's add it.

jontrulson, 2017-01-23:
I would need to experiment - using just a counter would be different for each platform.

jontrulson, 2017-03-13:
A fix was merged to MRAA that should fix this issue.  Now, there is a 5-second timeout enabled, so a broken circuit should no longer spin forever.

closed 2017-03-13 by jontrulson