eclipse-upm/upm #495: Version 1.0 not available on 01org

deadprogram, 2016-11-29:
The latest version of UPM is not available for installation on any of the 01org repos (https://download.01.org/iotgateway/) used for the Intel IoT Gateway, which is a problem for any code that needs to use the newer UPM APIs.

Propanu, 2016-12-03:
Hi @deadprogram, the latest and greatest should now be on there.

deadprogram, 2016-12-04:
I uninstalled the older versions, and installed using:

```
smart install https://download.01.org/iotgateway/rcpl21/x86_64/mraa-1.5.1-r0.2.corei7_64.rpm
smart install https://download.01.org/iotgateway/rcpl21/x86_64/upm-1.0.1-r0.0.corei7_64.rpm
```

However, none of my Python code will run. I receive  `ImportError: No module named upm` errors.

I looked in both `/usr/lib/python2.7/dist-packages` and also `/usr/lib/python2.7/site-packages` and did not find the upm python directory and packages I expected anywhere. What am I missing?

deadprogram, 2016-12-09:
This is now available thanks to the new upm 1.0.2 release. Thank you!

closed 2016-12-10 by jontrulson