eclipse-upm/upm #497: Undefined symbols in JS modules installed from npm with UPM 1.0.0-src

deadprogram, 2016-11-29:
Looks like there is a problem in some of the jsupm_* modules when using from JS for UPM version `1.0.0-src` when installed using npm.

When trying to call the jsupm_biss0001 module, even with just the normal example, I receive the following error:

```
ERROR: /usr/bin/node: symbol lookup error: /home/root/.node_app_slot/node_modules/jsupm_biss0001/build/Release/jsupm_biss0001.node: undefined symbol: _ZN3upm8BISS0001C1Ej
```

https://github.com/intel-iot-devkit/upm/issues/489 looks to be the same problem.

Propanu, 2016-11-29:
Yes, they are the same problem, and it looks like this happens in cases where we have a C++ source wrapping a C implementation. Need to investigate further to figure out what's causing it.

benjaminmathews, 2016-12-06:
Found the cause (and a solution) for the referenced issue : https://github.com/intel-iot-devkit/upm/issues/489
See this discussion: https://communities.intel.com/thread/108426
It wasn't a problem with the library. Code is working fine now, when we don't force Intel XDK to download and install the individual jsupm_* module.
Hope this info is of use.

Propanu, 2016-12-09:
On the Edison you'll have the libraries that provide these undefined symbols on the system's path, or the XDK will install them for you. I'm fairly certain that's why it works for #489, but that really hides the true issue, which has to do with how node-gyp compiles the sources from NPM on your system.

Propanu, 2016-12-15:
New version is up on NPM, 1.0.2-src. I've tracked the issue and found out it was caused by the makefile template node-gyp uses to build the sources. The target rules only use the base filename, thus if you have both a *module.c* and *module.cxx* in the same package, one of them will not be included in the build.

Since I doubt this can be fixed quickly by the node-gyp guys, I decided to rename our C sources while packing them in order to fix this. Everything seems to work as intended now.

closed 2017-01-23 by Propanu