eclipse-upm/upm #501: Building and running examples (for Java) is not clear

solidcell, 2016-12-05:
I'm not a Java programmer, so I'm sure that's part of it, but I feel that the docs are lacking.  At least in regards to Java.

I'm trying to follow the docs [here](https://github.com/intel-iot-devkit/upm/blob/master/docs/building.md) for building and running the examples, but after a few hours, I'm still not there.

I feel like a concrete example of how to use this would be helpful.  Here's what I've come up with so far:

Don't use the JDK that's already installed.  Instead:
```
wget --no-check-certificate --no-cookies - --header "Cookie: oraclelicense=accept-securebackup-cookie" "http://download.oracle.com/otn-pub/java/jdk/8u111-b14/jdk-8u111-linux-i586.tar.gz"
mkdir -p /usr/local
cd /usr/local
tar xzf ~/jdk-8u111-linux-i586.tar.gz
export JAVA_HOME=/usr/local/jdk1.8.0_111
mkdir -p /usr/local/bin
cd /usr/local/bin
ln -s $JAVA_HOME/bin/java
```

Then to get, say, the APA102 example to (*almost?*) run:
```
git clone repo https://github.com/intel-iot-devkit/upm.git
cd upm
mkdir build
cd build
cmake .. -DBUILDSWIGJAVA=ON -DBUILDEXAMPLES=ON
cd src/apa102
make
make install
cd ../../../examples/java
javac -cp /usr/local/lib/java/* APA102Sample.java
```

However, even at this point, it's not quite working:
```
root@edison:~/workspace/upm/examples/java# java APA102Sample
Exception in thread "main" java.lang.NoClassDefFoundError: upm_apa102/APA102
	at APA102Sample.main(APA102Sample.java:30)
Caused by: java.lang.ClassNotFoundException: upm_apa102.APA102
	at java.net.URLClassLoader.findClass(URLClassLoader.java:381)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:424)
	at sun.misc.Launcher$AppClassLoader.loadClass(Launcher.java:331)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:357)
	... 1 more
```

Again, I'm sure if I were a Java programmer, I would (probably?) have not had a problem with this, but I feel like it wouldn't hurt to help with a concrete example about how to actually run a Java example from this repo.  I've spent hours and gone through countless tutorials/articles, but still haven't gotten it to work.  Even [this seemingly informative article](http://shawnhymel.com/756/writing-your-own-upm-module-getting-started/) in the Java section says that it's not working.

Propanu, 2016-12-05:
Hi @solidcell, sorry to see you are having issues running the Java sensor examples, you're actually very close.

`CLASSPATH` is likely not set and hence why Java fails to load the library. You did figure out how to resolve this when compiling `javac -cp /usr/local/lib/java/* APA102Sample.java`, simply use the `-cp /usr/local/lib/java/*` option when running with `java` or set the environment variable beforehand. Keep in mind that the jars alone are not enough and `LD_LIBRARY_PATH` needs to be set to the location of `libjavaupm_apa102.so` (the one you compiled and installed).

Please note that Shawn's article is pretty old and our Java support has improved considerably since, especially when you use an IDE and a supported image on your device. For instance, with [Intel System Studio](https://software.intel.com/en-us/iot/tools-ide/ide/iss-iot-edition) you can create Java projects step by step, and we have several helpful samples in this [repo](https://github.com/intel-iot-devkit/how-to-code-samples). If you're using the IoT DevKit image consider updating the MRAA & UPM libraries with the built-in package manager, `opkg`, as described [here](https://software.intel.com/en-us/node/623543).

Propanu, 2016-12-05:
Oh, when running you'll also want the current directory added to the path since it has your APA102Sample class, thus something like:
`java -cp /usr/local/lib/java/upm_apa102.jar:. APA102Sample`

solidcell, 2016-12-06:
Thanks @Propanu, that worked perfectly :)

I don't have anything at `LD_LIBRARY_PATH`, but it seems to work fine with just doing `java -cp /usr/local/lib/java/upm_apa102.jar:. APA102Sample`.

As for the IDE, I like not relying on a specific IDE or too much magic.

closed 2016-12-06 by solidcell