eclipse-upm/upm #502: Multiple DS18B20 sensors with Intel Edison

benjaminmathews, 2016-12-06:
I'm using 
- Intel Edison with latest firmware, 
- Intel XDK for a Node.js project on the board
- upm module "jsupm_ds18b20" to communicate with the sensors
- mraa v1.5.1 and upm v1.0.1

Above setup works fine for a single sensor. But once I add a second sensor, it detects both, but is unable to read temperature from either sensor. i.e, the "ds18b20.devicesFound()" method returns "2", and I can read the ID's using "ds18b20.getId(0)" and "ds18b20.getId(1)". However, trying to update temperature readings using "ds18b20.update()" results in the following errors:
ERROR: readSingleTemp: crc check failed for device 0, returning previously measured temperature
ERROR: readSingleTemp: crc check failed for device 1, returning previously measured temperature
 
Connections are as detailed in : http://iotdk.intel.com/docs/master/upm/node/classes/ds18b20.html . "Data" pin of second sensor is connected directly to the data pin of the first. We're using only a single 1n4148 diode on the Tx line, which is common to both sensors, if that matters (though we tried it with separate circuits for each individual sensor too).
 
What am I missing here?

Propanu, 2016-12-09:
Perhaps @jontrulson can chime in on this, I never checked more than one sensor with this and the driver was recently changed to support additional features/devices.

jontrulson, 2016-12-09:
Hi, 

I have tried this sensor with a ds18b20 and a  ds2413 addressable switch, so I know in theory it can handle multiple devices.  The link you sent is unintelligible WRT the schematic, so here is one @Propanu  created:

[image]

Also, in my circuit I used a 4.7K ohm pull-up on each sensor.  I am also assuming that you are powering each sensor individually - ie: not using parasitic power.

One thing you could try is to see if you can update each sensor individually, such as:
`ds18b20.update(0)`
`ds18b20.update(1)`

It would be nice to know how wrong the CRC is, too bad I didn't add that code originally.  Otherwise, further investigation will be needed.

benjaminmathews, 2016-12-12:
[image]
yeah, that circuit is what we're using.
Individual sensor updates return the same error. The attached screenshot shows the exact console output. Those device ID's look a little weird too. Is it supposed to be like that?
 
If ds2413 is an addressable switch, wouldn't it use a different module/library? I mean something like "jsupm_ds2413", with a separate constructor and object and all that? Or doesn't that matter? I don't know, I'm just thinking aloud here... 

What other investigations should I try? How do I check how wrong the CRC is, if that would help?

jontrulson, 2016-12-12:
> Those device ID's look a little weird too. Is it supposed to be like that?

The Id's are 64bit binary values, it looks like they are (attempting) to be printed as utf-8

> If ds2413 is an addressable switch, wouldn't it use a different module/library? I mean something like "jsupm_ds2413", with a separate constructor and object and all that?

Yes it would - my point was that there should be no intrinsic problem in having multiple devices attached to the bus.  They should not interfere with each other.  To see what the CRC's are would require you to build the driver from scratch, adding some debugging information (like printing out the CRC's), or maybe bypassing the CRC check altogether (if the temperature data is still correct otherwise).  

If you have the C and C++ version of the ds18b20 examples present, or built, you might try those as well to see if you see the same issue.

Another odd thing, looking at the code:  In the contructor (DS18B20::DS18B20()), after the driver identifies how many devices are present, it specifically queries all of them to determine their temperature resolutions.  However, it does not check the CRC since we don't read the entire scratchpad into memory for this check.

It's very odd that this succeeds when a single device is present, but fails when 2 are present.  That should only happen if they are interfering with each other in some way.

ericksonla, 2016-12-22:
I am having this same problem- works fine with 1 sensor but crc check fails for 2 sensors. 

I also noticed behavior that seems wrong by physically unplugging the sensors. Using python:
`print(mysensobj.getTemperature(0), mysensobj.getTemperature(1))`
returns the following results:
```
readSingleTemp: crc check failed for device 0, returning previously measured temperature
0.0 0.0 #both sensors connected
21.0 0.0 #first sensor connected
21.0 0.0 #second sensor connected
```

The fact that its calling readSingleTemp suggests that maybe I should be using a different function to call multiple sensors?

Or maybe the fact that its reporting as sensor 0 either way suggests that its not correctly getting the ID input?

jontrulson, 2017-01-18:
Hi, I've investigated this problem and could reproduce it easily.  I have submitted PR #631 to MRAA that should fix this problem.  In short, a logic error was causing commands to be broadcast to all devices on the bus, rather than the specific device targeted by it's ROM ID.  

I've tested with 2 DS18B20's and they are working well now.  Thanks for reporting!

closed 2017-01-24 by Propanu

crprince, 2017-06-16:
Still having some issues with this if i let it run for a bit, even with a single sensor.  But not sure what to try debugging next. (Circuit as above, edison arduino breakout, reading every 5s, mraa/upm from java)

Seems to run fine for a few minutes, then an increasing number of errors. 
`
ds18b20_init: mraa_uart_ow_rom_search() failed, no devices detected
`
and 
```
readSingleTemp: crc check failed for device 0.  Got ff, expected c9. Returning
previously measured temperature
```
eventually stops reading altogether after a couple hours.

swapped out the hardware and tried the sensor on a Uno to see if it is faulty but works fine.

and thoughts would be appreciated.