eclipse-upm/upm #503: DFRobot SAINSMARTKS LCD cannot be used with MRAA subplatform

deadprogram, 2016-12-07:
I am unable to use the DFRobot SAINSMARTKS LCD display when using the MRAA Firmata subplatform, due to the pin naming convention of subplatforms (512 + the pin number) exceeding the allowed size of the params for the SAINSMART (uint8).

```
  File "/usr/lib64/python2.7/site-packages/upm/pyupm_i2clcd.py", line 1334, in __init__
    this = _pyupm_i2clcd.new_SAINSMARTKS(rs, enable, d0, d1, d2, d3, keypad)
NotImplementedError: Wrong number or type of arguments for overloaded function 'new_SAINSMARTKS'.
  Possible C/C++ prototypes are:
    upm::SAINSMARTKS::SAINSMARTKS(uint8_t,uint8_t,uint8_t,uint8_t,uint8_t,uint8_t,uint8_t)
    upm::SAINSMARTKS::SAINSMARTKS(uint8_t,uint8_t,uint8_t,uint8_t,uint8_t,uint8_t)
    upm::SAINSMARTKS::SAINSMARTKS(uint8_t,uint8_t,uint8_t,uint8_t,uint8_t)
    upm::SAINSMARTKS::SAINSMARTKS(uint8_t,uint8_t,uint8_t,uint8_t)
    upm::SAINSMARTKS::SAINSMARTKS(uint8_t,uint8_t,uint8_t)
    upm::SAINSMARTKS::SAINSMARTKS(uint8_t,uint8_t)
    upm::SAINSMARTKS::SAINSMARTKS(uint8_t)
    upm::SAINSMARTKS::SAINSMARTKS()
```

deadprogram, 2016-12-07:
I wonder if this is the same problem with https://github.com/intel-iot-devkit/upm/issues/498

deadprogram, 2016-12-09:
The new UPM 1.0.2 that was just released corrects this problem for me. Thank you!

Propanu, 2016-12-12:
Did this fix #498 for you as well?

deadprogram, 2016-12-14:
Yes it did. Both of these issues can now be closed, thanks.

closed 2016-12-14 by deadprogram