eclipse-upm/upm #508: LSM303 - Grove 6Axis V2.0 - Seeed

caymo04, 2016-12-26:
Good night!

I'm trying to connect Grove 6 Axis V2.0 of Seeed via I2C.
http://wiki.seeed.cc/Grove-6-Axis_AccelerometerAndCompass_V2.0/

INTEL GALILEO GEN 2 -- NodeJs 4.4.3 - MRAA 1.5.1.11 - UPM - 1.0.2.8

The error:
var myAccelrCompass = new accelrCompassSensor.LSM303(0);
                                        ^
Error: Illegal arguments for construction of _exports_LSM303
    at Error (native)
    at Object.<anonymous> (/home/root/.node_app_slot/main.js:30:23)
    at Module._compile (module.js:409:26)
    at Object.Module._extensions..js (module.js:416:10)
    at Module.load (module.js:343:32)
    at Function.Module._load (module.js:300:12)
    at Function.Module.runMain (module.js:441:10)
    at startup (node.js:139:18)
    at node.js:968:3

----------------------------------------------------

caymo04, 2016-12-27:
Good morning!

I tried also update jsump_lsm303.node file but, the process never starup :( ... same error "Illegal arguments for construction of _exports_LSM303)

Best regards and thanks a lot in advance.

Acaymo.

arfoll, 2016-12-27:
You seem to have updated node to v4.x however you need to recompile the UPM modules to match. Have you done so?

@intel-iot-devkit/upm-core-dev why is the lsm303 listed on the knownlimitations page with "fails to write to the configuration register properly and returns invalid data."? I'm pretty sure the sensor worked when I wrote that module, so is this on a particular board or is there a more global issue?

caymo04, 2016-12-28:
Hello @arfoll !!!

Thanks for your response.
UPM modules has been deleted and installed again.
How can I do a recompile?
I has installed NMP and, I has downloaded the last version of LSM303 via NMP
Remember, my sensor is 6-AXIS V2.
Best regards,

caymo04, 2016-12-29:
Hello again @arfoll 

When I try to re-compilation, the result it is:

root@galileo:~# **npm install jsupm_lsm303 -g**
|
> jsupm_lsm303@1.0.2-src install /usr/lib/node_modules/jsupm_lsm303
> node-gyp rebuild

make: Entering directory '/usr/lib/node_modules/jsupm_lsm303/build'
  CC(target) Release/obj.target/jsupm_lsm303/utilities/upm_utilities.o
  CXX(target) Release/obj.target/jsupm_lsm303/jsupm_lsm303JAVASCRIPT_wrap.o
../jsupm_lsm303JAVASCRIPT_wrap.cxx: In function 'SwigV8ReturnValue _wrap_new_int16Array(const SwigV8Arguments&)':
../jsupm_lsm303JAVASCRIPT_wrap.cxx:1841:79: warning: 'result' may be used uninitialized in this function [-Wmaybe-uninitialized]
   SWIGV8_SetPrivateData(self, result, SWIGTYPE_p_int16Array, SWIG_POINTER_OWN);
                                                                               ^
../jsupm_lsm303JAVASCRIPT_wrap.cxx: In function 'SwigV8ReturnValue _wrap_LSM303_getAccelZ(const SwigV8Arguments&)':
../jsupm_lsm303JAVASCRIPT_wrap.cxx:847:78: warning: 'result' may be used uninitialized in this function [-Wmaybe-uninitialized]
 #define SWIGV8_NUMBER_NEW(num) v8::Number::New(v8::Isolate::GetCurrent(), num)
                                                                              ^
../jsupm_lsm303JAVASCRIPT_wrap.cxx:3849:11: note: 'result' was declared here
   int16_t result;
           ^
../jsupm_lsm303JAVASCRIPT_wrap.cxx: In function 'SwigV8ReturnValue _wrap_LSM303_getAccelY(const SwigV8Arguments&)':
../jsupm_lsm303JAVASCRIPT_wrap.cxx:847:78: warning: 'result' may be used uninitialized in this function [-Wmaybe-uninitialized]
 #define SWIGV8_NUMBER_NEW(num) v8::Number::New(v8::Isolate::GetCurrent(), num)
                                                                              ^
../jsupm_lsm303JAVASCRIPT_wrap.cxx:3766:11: note: 'result' was declared here
   int16_t result;
           ^
../jsupm_lsm303JAVASCRIPT_wrap.cxx: In function 'SwigV8ReturnValue _wrap_LSM303_getAccelX(const SwigV8Arguments&)':
../jsupm_lsm303JAVASCRIPT_wrap.cxx:847:78: warning: 'result' may be used uninitialized in this function [-Wmaybe-uninitialized]
 #define SWIGV8_NUMBER_NEW(num) v8::Number::New(v8::Isolate::GetCurrent(), num)
                                                                              ^
../jsupm_lsm303JAVASCRIPT_wrap.cxx:3683:11: note: 'result' was declared here
   int16_t result;
           ^
../jsupm_lsm303JAVASCRIPT_wrap.cxx: In function 'SwigV8ReturnValue _wrap_LSM303_getCoorZ(const SwigV8Arguments&)':
../jsupm_lsm303JAVASCRIPT_wrap.cxx:847:78: warning: 'result' may be used uninitialized in this function [-Wmaybe-uninitialized]
 #define SWIGV8_NUMBER_NEW(num) v8::Number::New(v8::Isolate::GetCurrent(), num)
                                                                              ^
../jsupm_lsm303JAVASCRIPT_wrap.cxx:3515:11: note: 'result' was declared here
   int16_t result;
           ^
../jsupm_lsm303JAVASCRIPT_wrap.cxx: In function 'SwigV8ReturnValue _wrap_LSM303_getCoorY(const SwigV8Arguments&)':
../jsupm_lsm303JAVASCRIPT_wrap.cxx:847:78: warning: 'result' may be used uninitialized in this function [-Wmaybe-uninitialized]
 #define SWIGV8_NUMBER_NEW(num) v8::Number::New(v8::Isolate::GetCurrent(), num)
                                                                              ^
../jsupm_lsm303JAVASCRIPT_wrap.cxx:3432:11: note: 'result' was declared here
   int16_t result;
           ^
../jsupm_lsm303JAVASCRIPT_wrap.cxx: In function 'SwigV8ReturnValue _wrap_LSM303_getCoorX(const SwigV8Arguments&)':
../jsupm_lsm303JAVASCRIPT_wrap.cxx:847:78: warning: 'result' may be used uninitialized in this function [-Wmaybe-uninitialized]
 #define SWIGV8_NUMBER_NEW(num) v8::Number::New(v8::Isolate::GetCurrent(), num)
                                                                              ^
../jsupm_lsm303JAVASCRIPT_wrap.cxx:3349:11: note: 'result' was declared here
   int16_t result;
           ^
../jsupm_lsm303JAVASCRIPT_wrap.cxx: In function 'SwigV8ReturnValue _wrap_LSM303_getAcceleration(const SwigV8Arguments&)':
../jsupm_lsm303JAVASCRIPT_wrap.cxx:3246:41: warning: 'result' may be used uninitialized in this function [-Wmaybe-uninitialized]
   jsresult = SWIG_From_int((int)(result));
                                         ^
../jsupm_lsm303JAVASCRIPT_wrap.cxx: In function 'SwigV8ReturnValue _wrap_LSM303_getCoordinates(const SwigV8Arguments&)':
../jsupm_lsm303JAVASCRIPT_wrap.cxx:3163:41: warning: 'result' may be used uninitialized in this function [-Wmaybe-uninitialized]
   jsresult = SWIG_From_int((int)(result));
                                         ^
../jsupm_lsm303JAVASCRIPT_wrap.cxx: In function 'SwigV8ReturnValue _wrap_LSM303_getHeading(const SwigV8Arguments&)':
../jsupm_lsm303JAVASCRIPT_wrap.cxx:1683:34: warning: 'result' may be used uninitialized in this function [-Wmaybe-uninitialized]
   return SWIG_From_double  (value);
                                  ^
[... 6 lines cut ...]
                                                                                ^
../jsupm_lsm303JAVASCRIPT_wrap.cxx: In function 'SwigV8ReturnValue _wrap_new_LSM303__SWIG_2(const SwigV8Arguments&, V8ErrorHandler&)':
../jsupm_lsm303JAVASCRIPT_wrap.cxx:2839:80: warning: 'result' may be used uninitialized in this function [-Wmaybe-uninitialized]
   SWIGV8_SetPrivateData(self, result, SWIGTYPE_p_upm__LSM303, SWIG_POINTER_OWN);
                                                                                ^
[... cut ...]
   SWIGV8_SetPrivateData(self, result, SWIGTYPE_p_upm__LSM303, SWIG_POINTER_OWN);
                                                                                ^
../jsupm_lsm303JAVASCRIPT_wrap.cxx:2561:16: note: 'result' was declared here
   upm::LSM303 *result;
                ^
  CXX(target) Release/obj.target/jsupm_lsm303/lsm303.o
  SOLINK_MODULE(target) Release/obj.target/jsupm_lsm303.node
  COPY Release/jsupm_lsm303.node
make: Leaving directory '/usr/lib/node_modules/jsupm_lsm303/build'
-----------------------------------------------------------

Regards!

arfoll, 2017-01-02:
Make sure you are loading the right module, I'm guessing there is another jsupm_lsm303 module that is being loaded? Check the upm ipk package and what/where it's providing it's stuff

Propanu, 2017-01-02:
Hi, the known limitation applies to the Edison with the Arduino breakout as far as I recall. On the Galileo Gen 2 this should work fine, just keep in mind that NPM installs under `/usr/local/lib/node_modules` and the Galileo image you are using might already have the old libraries under `/usr/local/node_modules`. I think Node.js can be forced to use the new one by setting your `NODE_PATH` environment variable.

caymo04, 2017-01-04:
Thanks @arfoll and @Propanu 

My test has begun:

 - Download image linux from Intel website
 - Update to MRAA 1.5.1.8 and UPM 1.0.2.8
 - Intel XDK
 - Same code with LSM303
 - Running --- > Fail!

ok, I try to install jsupm_lsm303 from NPM. Under /usr/lib directory, call to ->  npm install jsupm_lsm303 -g 

The result is: No installed correctly and same error "Error: Illegal arguments for construction of _exports_LSM303"

My question is:

 - Is necesary update another library?

Best regards and thanks a lot in advance.

Acaymo.

arfoll, 2017-01-04:
do an `ls -l /usr/lib/node_modules/` and `ls -ls /usr/local/lib/node_modules`. I think you'll find you have two versions. Going inside /usr/lib and doing an `npm install -g` makes no difference, you'd have to do a non `-g` install for that to work...

caymo04, 2017-01-04:
@arfoll 

I has tested the directory and, I don't have two versions.
Also, I has reinstalled and execute into /usr/lib the comand -> npm install jsump_lsm303
And same error.

The I2Cdetect is detecting the 6Axis sensor:
root@galileo:~/.node_app_slot# i2cdetect -y -r 0

     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- 1e --
20: -- -- -- -- -- UU UU UU -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- UU -- -- -- -- -- -- -- --
50: -- -- -- -- UU UU UU UU -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: 70 -- -- -- -- -- -- --

 and my program in Node.Js is:

/*jslint node:true, vars:true, bitwise:true, unparam:true */
/*jshint unused:true */
/*
* Author: Zion Orent
* Copyright (c) 2014 Intel Corporation.
*
* Permission is hereby granted, free of charge, to any person obtaining
* a copy of this software and associated documentation files (the
* "Software"), to deal in the Software without restriction, including
* without limitation the rights to use, copy, modify, merge, publish,
* distribute, sublicense, and/or sell copies of the Software, and to
* permit persons to whom the Software is furnished to do so, subject to
* the following conditions:
*
* The above copyright notice and this permission notice shall be
* included in all copies or substantial portions of the Software.
*
* THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND,
* EXPRESS OR IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF
* MERCHANTABILITY, FITNESS FOR A PARTICULAR PURPOSE AND
* NONINFRINGEMENT. IN NO EVENT SHALL THE AUTHORS OR COPYRIGHT HOLDERS BE
* LIABLE FOR ANY CLAIM, DAMAGES OR OTHER LIABILITY, WHETHER IN AN ACTION
* OF CONTRACT, TORT OR OTHERWISE, ARISING FROM, OUT OF OR IN CONNECTION
* WITH THE SOFTWARE OR THE USE OR OTHER DEALINGS IN THE SOFTWARE.
*/

var accelrCompassSensor = require('jsupm_lsm303');
// Instantiate LSM303 compass on I2C

var myAccelrCompass = new accelrCompassSensor.LSM303(0);

var successFail, coords, outputStr, accel;
var myInterval = setInterval(function()
{
        // Load coordinates into LSM303 object
        successFail = myAccelrCompass.getCoordinates();
        // in XYZ order. The sensor returns XZY,
        // but the driver compensates and makes it XYZ
        coords = myAccelrCompass.getRawCoorData();

    // Print out the X, Y, and Z coordinate data using two different methods
        outputStr = "coor: rX " + coords.getitem(0)
                                                                                                                                                     
    // Print out the X, Y, and Z coordinate data using two different methods
        outputStr = "coor: rX " + coords.getitem(0)
                                        + " - rY " + coords.getitem(1)
                                        + " - rZ " + coords.getitem(2);
        console.log(outputStr);
        outputStr = "coor: gX " + myAccelrCompass.getCoorX()
                                + " - gY " + myAccelrCompass.getCoorY()
                                + " - gZ " + myAccelrCompass.getCoorZ();
        console.log(outputStr);

    // Get and print out the heading
        console.log("heading: " + myAccelrCompass.getHeading());

    // Get the acceleration
        myAccelrCompass.getAcceleration();
        accel = myAccelrCompass.getRawAccelData();
    // Print out the X, Y, and Z acceleration data using two different methods
        outputStr = "acc: rX " + accel.getitem(0)
                                + " - rY " + accel.getitem(1)
                                + " - Z " + accel.getitem(2);
        console.log(outputStr);
        outputStr = "acc: gX " + myAccelrCompass.getAccelX()
                                + " - gY " + myAccelrCompass.getAccelY()
                                + " - gZ " + myAccelrCompass.getAccelZ();
        console.log(outputStr);
        console.log(" ");
}, 1000);

// Print message when exiting
process.on('SIGINT', function()
{
        clearInterval(myInterval);
        myAccelrCompass = null;
        accelrCompassSensor.cleanUp();
        accelrCompassSensor = null;
        console.log("Exiting");
        process.exit(0);
});

caymo04, 2017-01-04:
@arfoll @Propanu 

Please, see bold text from this response of "journalctl -f" command:

root@galileo:/usr/lib/node_modules/jsupm_lsm303/build/Release# journalctl -f
-- Logs begin at Mon 2001-01-01 00:00:23 UTC. --
Jan 04 21:04:23 galileo xdk-daemon[1333]: at Object.Module._extensions..js (module.js:416:10)
Jan 04 21:04:23 galileo xdk-daemon[1333]: at Module.load (module.js:343:32)
Jan 04 21:04:23 galileo xdk-daemon[1333]: at Function.Module._load (module.js:300:12)
Jan 04 21:04:23 galileo xdk-daemon[1333]: at Function.Module.runMain (module.js:441:10)
Jan 04 21:04:23 galileo xdk-daemon[1333]: at startup (node.js:139:18)
Jan 04 21:04:23 galileo xdk-daemon[1333]: at node.js:968:3
Jan 04 21:04:23 galileo xdk-daemon[1333]: child exited
Jan 04 21:04:23 galileo xdk-daemon[1333]: Child process closed
Jan 04 21:05:01 galileo systemd-timesyncd[117]: interval/delta/delay/jitter/drift 128s/+0.001s/0.060s/0.315s/-54ppm
Jan 04 21:07:09 galileo systemd-timesyncd[117]: interval/delta/delay/jitter/drift 256s/+0.001s/0.063s/0.311s/-52ppm
Jan 04 21:11:09 galileo xdk-daemon[1333]: received: {"channel":"command","message":"run","arguments":[]}
Jan 04 21:11:09 galileo xdk-daemon[1333]: V8PORT REC
**Jan 04 21:11:14 galileo libmraa[1461]: libmraa version v1.5.1 initialised by user 'root' with EUID 0
Jan 04 21:11:14 galileo libmraa[1461]: libmraa initialised for platform 'Intel Galileo Gen 2' of type 1
Jan 04 21:11:14 galileo libmraa[1461]: i2c_init: Selected bus 0
Jan 04 21:11:14 galileo libmraa[1461]: i2c0: write: Access error: Remote I/O error**
Jan 04 21:11:14 galileo xdk-daemon[1333]: stderr: /home/root/.node_app_slot/main.js:5
Jan 04 21:11:14 galileo xdk-daemon[1333]: stderr: var myAccelrCompass = new accelrCompassSensor.LSM303(0);
Jan 04 21:11:14 galileo xdk-daemon[1333]: ^
Jan 04 21:11:14 galileo xdk-daemon[1333]: Error: Illegal arguments for construction of _exports_LSM303
Jan 04 21:11:14 galileo xdk-daemon[1333]: at Error (native)
Jan 04 21:11:14 galileo xdk-daemon[1333]: at Object.<anonymous> (/home/root/.node_app_slot/main.js:5:23)
Jan 04 21:11:14 galileo xdk-daemon[1333]: at Module._compile (module.js:409:26)
Jan 04 21:11:14 galileo xdk-daemon[1333]: at Object.Module._extensions..js (module.js:416:10)
Jan 04 21:11:14 galileo xdk-daemon[1333]: at Module.load (module.js:343:32)
Jan 04 21:11:14 galileo xdk-daemon[1333]: at Function.Module._load (module.js:300:12)
Jan 04 21:11:14 galileo xdk-daemon[1333]: at Function.Module.runMain (module.js:441:10)
Jan 04 21:11:14 galileo xdk-daemon[1333]: at startup (node.js:139:18)
Jan 04 21:11:14 galileo xdk-daemon[1333]: at node.js:968:3
Jan 04 21:11:14 galileo xdk-daemon[1333]: child exited
Jan 04 21:11:14 galileo xdk-daemon[1333]: Child process closed

------------------------------------------------

Regards!

caymo04, 2017-01-04:
Solved!   @arfoll @Propanu 

When I checked the library .cxx I can see:

   m_addrMag = addrMag;
    m_addrAcc = addrAcc;

    // 0x27 is the 'normal' mode with X/Y/Z enable
    setRegisterSafe(m_addrAcc, CTRL_REG1_A, 0x27);

-----------------------
But, the addreAcc it is not valid!!! 
I tried with this parameters when I request the module --> LSM303(0,0x1e,0x1e,8); and EUREKA!:

coor: rX 0 - rY 0 - rZ 0
coor: gX 0 - gY 0 - gZ 0
heading: 0
acc: rX -483 - rY 386 - Z 17446
acc: gX -483 - gY 386 - gZ 17446

At now, I need activate another parameters but, this is another history.

Best regards for the support 👍 

Acaymo.

closed 2017-01-04 by caymo04

Propanu, 2017-01-05:
Hi @caymo04, glad you figured it out, however there's a bit more to it. The Grove 6-Axis V2.0 uses the newer LSM303D chip instead of the LSM303DLHC which was used in V1 (and to write the UPM driver). The major difference is that LSM303D uses a single address for both acceleration and compass data (hence why it worked with 0x1e, 0x1e) but you won't get any valid data for the magnetometer.

I think this needs to be made more clear in our documentation and personally, I would derive a new LSM303D class from LSM303 to provide the extra functionality.

I wouldn't mind one bit if you want to open another issue asking for proper LSM303D support.

caymo04, 2017-01-05:
Hi @Propanu , is true!

I will opening a new issue about LSM303D.

Thanks for your comments