eclipse-upm/upm #509: How to I use mraa/upm(smartdrive) in intel edison with android thingd SO?

neuberfran, 2016-12-27:
I know about this https://developer.android.com/things/sdk/pio/i2c.html
But if I can use a UPM sensor (smartdrive in my case) I will can to abstract enough development work
https://github.com/intel-iot-devkit/upm/tree/master/src/smartdrive

Propanu, 2017-01-02:
Hi @neuberfran thanks for asking. MRAA & UPM support for Android Things is being worked on and will become available soon for our Java bindings through Android Studio.

Propanu, 2017-05-11:
We have officially released MRAA & UPM for Android Things! You can read more about it [here](https://software.intel.com/en-us/blogs/2017/04/26/upm-and-mraa-support-for-android-things).

neuberfran, 2017-05-13:
Hi help me with
this in smartdrive UPM https://drive.google.com/file/d/0B9gPdWq4OH-XX2lJb2pvUGRrZHM/view?usp=sharing

https://github.com/intel-iot-devkit/upm/issues/461

Please change https://github.com/intel-iot-devkit/upm/blob/master/src/smartdrive/smartdrive.cxx and https://github.com/intel-iot-devkit/upm/blob/master/src/smartdrive/smartdrive.hpp for this

https://github.com/neuberfran/SmartDriveCorrect/blob/master/smartdrive.cxx and this https://github.com/neuberfran/SmartDriveCorrect/blob/master/smartdrive.hpp

**_I did several tests and I advise that the best thing to do is_**

### **_I need this urgent to put upm-smartdrive-1.2.0/ in gradle_**

pylbert, 2017-05-15:
This artifact is now available in JCenter.  For gradle you can use a compile dependency:

compile 'io.mraa.at.upm:upm_smartdrive:1.3.0'

neuberfran, 2017-05-16:
I don't understood. It would be a lot easier for you to make the changes I'm asking for.

jontrulson, 2017-05-16:
Hi, could you just submit a proper PR for this if there are problems with it?

neuberfran, 2017-05-16:
How to I do this (PR)? Why do not you do this to me?

jontrulson, 2017-05-16:
A PR is a Pull Request.  That's when somebody has a patch to add some feature or fix bugs in existing code, then they create a PR so it can be reviewed and eventually merged.

I can't speak for others, but I simply cannot go looking through someone's git repo to try and figure out what is changed and create the PR myself.

PR's are discussed here: https://help.github.com/articles/about-pull-requests/

If there is a problem with the smartdrive, and you have fixed it, then you should create a PR that contains the fixes and submit it so it can be looked at and possibly merged into UPM.

neuberfran, 2017-05-16:
Thank you. I had already done something like that and did it again now

https://github.com/intel-iot-devkit/upm/pull/551

neuberfran, 2017-05-16:
+jontrulson I need your help. I have issues with a PR create

Look this https://github.com/intel-iot-devkit/upm/pull/551 please

neuberfran, 2017-05-17:
+jontrulson 

How to I make whatever changes to the smartdrive src code needed to make it work in my (computer) branch repository?

I created in c:/ two pastes . In paste errataupm I have files with erros. In paste corretaupm I have files corrects

neuberfran, 2017-05-17:
I have errors with git push -u origin smartdrive-fixes

https://drive.google.com/file/d/0B9gPdWq4OH-XMUx1N3VFYlBIM3c/view?usp=sharing