eclipse-upm/upm #511: "UPM: Unknown Platform!" error when compiling C++ with -std=c++1y flag on Windriver Linux

deadprogram, 2017-01-10:
When attempting to compile any C++ code that uses UPM and also the `-std=c++1y` flag on Windriver Linux, the compilation fails with an "UPM: Unknown Platform!" error.

Here is the call stack:

```
In file included from /usr/include/upm/upm_math.h:28:0,
                 from /usr/include/upm/upm.h:36,
                 from /usr/include/upm/jhd1313m1.h:33,
                 from /usr/include/upm/jhd1313m1.hpp:37,
                 from ../src/temperature-lcd.cpp:48:
/usr/include/upm/upm_platform.h:39:3: error: #error "UPM: Unknown Platform!"
```

In addition, there are also some further errors:

```
/usr/include/upm/upm_utilities.h:92:21: error: variable or field 'upm_clock_init' declared void
 void upm_clock_init(upm_clock_t *clock);
                     ^
/usr/include/upm/upm_utilities.h:92:21: error: 'upm_clock_t' was not declared in this scope
/usr/include/upm/upm_utilities.h:102:25: error: 'upm_clock_t' was not declared in this scope
 uint32_t upm_elapsed_ms(upm_clock_t *clock);
                         ^
/usr/include/upm/upm_utilities.h:112:25: error: 'upm_clock_t' was not declared in this scope
 uint32_t upm_elapsed_us(upm_clock_t *clock);
```

I do not know if those are due to the first error, or are errors in their own right.

jontrulson, 2017-01-10:
Could you edit include/upm_platform.h and change line 32 from:

`#if defined(linux)`

to

`#if defined(__linux__)`

(ie: linux preceded and followed by two underscores) and see if that works for you?

deadprogram, 2017-01-10:
I presume you mean on the matching Docker instance, correct?

jontrulson, 2017-01-10:
I mean in whatever source tree/environment you are building upm in.

If it's easier, I can merge this fix into UPM master, and you can go from there.

@pylbert and I had discussed this issue a couple months ago, I had mistakenly assumed a patch had already been merged to fix this.

Propanu, 2017-01-10:
@jontrulson please do so.

jontrulson, 2017-01-10:
Merged as commit 6ea65a16a48819ab8aa82293adc5ed537bc63370

deadprogram, 2017-01-10:
That change allows the project to build. Thanks, closing.

deadprogram, 2017-01-31:
Any idea when this is going to be released? It is a blocker for devs using the latest ISS and toolchain.

Propanu, 2017-02-02:
We need to tag before we can push this to 01.org, I'll take care of it.

Propanu, 2017-09-22:
Should no longer be an issue as new UPM versions have been released on 01.org. Closing.

closed 2017-09-22 by Propanu