eclipse-upm/upm #512: [Python] Water Flow Sensor (WFS) Throws a Segmentation Fault when MRAA Import Present

thoughtentity, 2017-01-21:
## [Python] Water Flow Sensor (WFS) Throws a Segmentation Fault when MRAA Import Present

### Environment Information
- Hardware: Intel Edison
- Python Version: 2.7.3
- UPM Version: 1.0.2

### Problem

When using the UPM Water Flow Sensor (WFS) in a Python application on the Intel Edision the sensor will throw a Segmentation Fault on pulse when the MRAA library is imported into the global Python module scope.

### Reproduction

The following minimum Python file will reproduce the issue. Removing the MRAA import will allow the Python file to run as expected.

> wfs.py

```Python
#!/usr/bin/python

from __future__ import print_function
import time, sys, signal, atexit
from upm import pyupm_wfs as upmwfs
from mraa import Gpio

def main():
    # Instantiate a Water Flow Sensor on digital pin D2
    myWaterFlow = upmwfs.WFS(2)

    ## Exit handlers ##
    # This stops python from printing a stacktrace when you hit control-C
    def SIGINTHandler(signum, frame):
        raise SystemExit

    # This function lets you run code on exit,
    # including functions from myWaterFlow
    def exitHandler():
        myWaterFlow.stopFlowCounter()
        print("Exiting")
        sys.exit(0)

    # Register exit handlers
    atexit.register(exitHandler)
    signal.signal(signal.SIGINT, SIGINTHandler)

    # set the flow counter to 0 and start counting
    myWaterFlow.clearFlowCounter()
    myWaterFlow.startFlowCounter()

    while (1):
        # we grab these (millis and flowCount) just for display
        # purposes in this example
        millis = myWaterFlow.getMillis()
        flowCount = myWaterFlow.flowCounter()

        fr = myWaterFlow.flowRate()

        # output milliseconds passed, flow count, and computed flow rate
        outputStr = "Millis: {0} Flow Count: {1} Flow Rate: {2} LPM".format(
        millis, flowCount, fr)
        print(outputStr)
        time.sleep(2)

if __name__ == '__main__':
    main()
```
### Expected Result

When the Python file is run the WFS should give the proper water flow reading without throwing a segmentation fault when the sensor pulses. Reguardless of the presense of the MRAA import.

Propanu, 2017-01-23:
Hi, can you try putting `from mraa import Gpio` before the upm import? I think there was a similar issue where loading MRAA and UPM at the same time would break in node.js.

Propanu, 2017-05-11:
@thoughtentity any update on this?