eclipse-upm/upm #513: "error: implicit declaration of function ‘usleep’" on debian jessie

vadimadr, 2017-01-23:
I am trying to build upm on debian jessie. Mraa is compiled and installed successfully. However, when I try to build upm I get the following error:
```
[  2%] Building C object src/utilities/CMakeFiles/utilities-c.dir/upm_utilities.c.o
/upm-src/src/utilities/upm_utilities.c: In function ‘upm_delay_ms’:
/upm-src/src/utilities/upm_utilities.c:66:5: error: implicit declaration of function ‘usleep’ [-Werror=implicit-function-declaration]
     usleep(1000 * time);
     ^
/upm-src/src/utilities/upm_utilities.c: At top level:
cc1: error: unrecognized command line option "-Wno-misleading-indentation" [-Werror]
cc1: all warnings being treated as errors
make[2]: *** [src/utilities/CMakeFiles/utilities-c.dir/upm_utilities.c.o] Error 
```
```
cmake version 3.0.2
gcc version 4.9.2 (Debian 4.9.2-10) 
```

Propanu, 2017-01-23:
My quick guess is that `#if defined(__linux__)` doesn't work on Debian as intended. Can you try building with `make UPM_PLATFORM_LINUX=1` and let us know if that works for you?

vadimadr, 2017-01-24:
Unfortunately, same error.

vadimadr, 2017-01-24:
If you have docker you can reproduce this error yourself with [this dockerfile](https://gist.github.com/vadimadr/36375707e5591c32c0d8c8b93fee9a44)

Download it and run: `docker build -t python-mraa-upm . `

Propanu, 2017-02-01:
We'll provide a nanosleep implementation for the delay function instead, to match the standard. Also, it might be worth giving it a try with CMake 3.1+, as it detects the standard to use a bit differently. For older CMake we set `-std=c++11`, but on Debian you might want to use `-std=gnu++11` instead. The `CMAKE_CXX_FLAGS` variable can be used with CMake for that.

jontrulson, 2017-02-02:
Some fixes have been committed that use nanosleep() instead of usleep() for delays, and also defines the appropriate POSIX_C_SOURCE standard in upm_utilities.c needed to avoid these warnings/errors.

I have tested with your Dockerfile and current master should build without issues now.

Thanks for supplying the Dockerfile to reproduce this issue!

vadimadr, 2017-02-04:
@jontrulson Thank you, now it works. I also created [repository in docker hub](https://hub.docker.com/r/vadimadr/docker-python-iot-mock/) for python with mraa and upm included.

closed 2017-02-04 by vadimadr

kynan, 2017-03-19:
Unfortunately this issue has come back:
```
[Build]    /usr/src/upm-master/src/p9813/p9813.c: In function ‘p9813_send_byte’:
[Build]    /usr/src/upm-master/src/p9813/p9813.c:103:9: error: implicit declaration of function ‘usleep’ [-Werror=implicit-function-declaration]
[Build]             usleep(20);
[Build]             ^
[Build]    /usr/src/upm-master/src/p9813/p9813.c: At top level:
[Build]    cc1: error: unrecognized command line option "-Wno-misleading-indentation" [-Werror]
[Build]    cc1: all warnings being treated as errors
```

jontrulson, 2017-03-20:
Since you are building from master, these kinds of problems may crop up from time to time.  To allow you to build, you might want to disable warnings-as-error - add 

`-DWERROR=NO`

To your cmake command line.