eclipse-upm/upm #518: JAVA modules missing implementations of certain methods

pylbert, 2017-02-01:
The ${module_src} is no longer built into each wrapper module (python, JAVA, and NodeJS).  Python and NodeJS are fine with this however JAVA wrappers require functionality added with #ifdefs in the cxx src.

Example (bma220):

```
bma220/bma220.hpp:

  #if defined(SWIGJAVA) || defined(JAVACALLBACK)
      void installISR(int gpio, mraa::Edge level, jobject runnable);
  #else
```

```
bma220/bma220.cxx:
#if defined(SWIGJAVA) || (JAVACALLBACK)
void BMA220::installISR(int gpio, mraa::Edge level, jobject runnable)
{
...
```

The bma220 library target creates a upm-bma220.so library w/o the above method.
The javaupm_bma220 target then builds a wrapper library (libjavaupm_bma220.so) and includes the definition of the above method (from the header's ifdefs).  At this point, libjavaupm_bma220.so incorrectly assumes the implementation for the above method is in the upm-bma220.so library.

jontrulson, 2017-02-01:
Yes, I brought this up a few months ago when the build system was overhauled.  I referred to it as "The Java Problem" :)

Python and JS are fine with this since they don't need the special sauce that Java does in order to handle interrupts.  We can also "handle" pointers in JS/Python - we cannot in Java.

A solution I proposed, was to move the JAVA specific implementations of these methods out of the .cxx files and put them in the .hpp files (which are properly processed with the current building scheme).

pylbert, 2017-02-01:
I would like to do a temp fix for this - add the src back for the java swig module.

Some of these could be cleaned up by moving the Callback class seen in various libraries out to a C++ header for use by all.  Access via directors could be provided to this Callback class from a global java.i file.  Others such as the runnable objects could be handled entirely in the swig interface file since they are only used by java.

jontrulson, 2017-02-01:
Hmm.  I think you tested the use of directors in this fashion to try to resolve this problem (at least for interrupts), and the result was a 5x slowdown in interrupt performance... Or am I mis-remembering?

Implementing this stuff fully in the driver's Java interface file might also be an option.

Though personally, I'd prefer to keep it in the .hpp file for documentation and "understandability" reasons.

jontrulson, 2017-02-02:
Oh, and I'm okay with your intermediate fix for now.  We should keep this issue open so that eventually when all of the drivers suffering from this issue are fixed, we can then remove it.

pylbert, 2017-02-02:
A temporary fix will provide original functionality while we come up with a complete solution.

A handful of these use directors over runnables so I assume whoever authored them was satisfied with the functionality.  I gave up on converting a runnable to a director approach because of a speed difference.  This proposal is separate from that.  Looking through the SWIGJAVA defines I see this used for more than just callbacks.  

This for example...

```
./h3lis331dl/h3lis331dl.cxx:#ifdef SWIGJAVA
./h3lis331dl/h3lis331dl.cxx-float *H3LIS331DL::getAcceleration()
./h3lis331dl/h3lis331dl.cxx-{
./h3lis331dl/h3lis331dl.cxx-  float *v = new float[3];
./h3lis331dl/h3lis331dl.cxx-  getAcceleration(&v[0], &v[1], &v[2]);
./h3lis331dl/h3lis331dl.cxx-  return v;

```

is not ideal IMO.  A define was placed the the base library implementation which adds a C++ method that happily lets a caller shoot themselves in the foot WRT leaking memory.  I guess this makes it easier for use in JAVA?  We should look into ways of making this easier to use in the wrappers that don't compromise the original C++.  Possibly:

 ` std::vector<float> H3LIS331DL::getAcceleration();
`

I see you changed some of the methods which do not have JAVA ifdef guards:
```

./mma7361/mma7361.cxx-float *MMA7361::getAcceleration()
./mma7361/mma7361.cxx-{
./mma7361/mma7361.cxx:  static float data[3];
./mma7361/mma7361.cxx-
./mma7361/mma7361.cxx-  getAcceleration(&data[0], &data[1], &data[2]);
./mma7361/mma7361.cxx-
./mma7361/mma7361.cxx-  return data;
./mma7361/mma7361.cxx-}

```
which, among other things, does not seem threadsafe although it mitigates leaking memory.  I think we need a good solution which solves the need of all the JAVA defines in the UPM C++.

pylbert, 2017-02-02:
Reopening so this can be tracked.

jontrulson, 2017-02-02:
Yes, we did talk about these cases already.  In the first case, if you look at the implementation in the Java interface file, you will see that the memory allocated is indeed freed properly.

As for the second example, yes, you have a point there.  Shame on me.

We have also talked about getting rid of this kind of stuff in favor of making use of std::vector.  A change I am all in favor of, but we need to spend some time to actually try it and see how well it will work.

jontrulson, 2017-03-08:
If you look at commit d4b536b (at least the interesting parts), I am now using std::vector and a new upm_vectortypes.i interface file to return array data for Java (and everything else) instead of array pointers.  In addition, ISR handling for Java is implemented in the .hpp file now - so for at least that driver, no more "special" processing of Java would be required on the .cxx file.  This is what I will be doing going forward.

We should have been using std::vectors a long time ago :)  The resulting code (both actual, and SWIG interface) is much cleaner, and the SWIG languages can use them in a much more "natural" (to the language) way.  Check out the JavaScript, Java, and Python example changes.

commit cdb75e8c (2017-02-02): swig_java: Fix for #518 - java wrapper implementations

This commit fixes #518 by adding source back into the CXX swig-
generated wrapper.