eclipse-upm/upm #522: Raspberry pi Installation

efbick, 2017-02-10:
I am having trouble building and installing upm on raspbian. I'm thinking its because im missing a few modules but I couldn't find some of them to install.

`pi@raspberrypi:~/Documents/IoT/upm/build $ cmake ..
-- Warnings as errors enabled (-Werror), disable with -DWERROR=off
-- Checking for module 'mraa>=1.4.0'
--   Found mraa, version 1.5.1
-- Checking for module 'libbacnet'
--   No package 'libbacnet' found
-- Checking for module 'libmodbus>=3.1.2'
--   
-- Checking for module 'libopenzwave'
--   No package 'libopenzwave' found
-- Could NOT find JPEG (missing:  JPEG_LIBRARY JPEG_INCLUDE_DIR) 
-- Node version is 4.7.3
-- Node using v8 4.5.103.43
-- Building python2 modules with python-2.7.9
-- Building python3 modules with python-3.4.2
CMake Warning at CMakeLists.txt:201 (message):
  Failed to retrieve UPM version with 'git describe' (using v1.0.2).  Check
  that git is installed and this is a valid git repo.

-- UPM Version v1.0.2
-- Target arch is armv6
-- BUILDDOC=OFF, python modules will not contain cxx documentation
-- Generating pkg-config files for 259 libraries...
-- 	Found C   libraries: 71
-- 	Found CXX libraries: 188
-- Generating pkg-config files for 259 libraries... Complete
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/Documents/IoT/upm/build
pi@raspberrypi:~/Documents/IoT/upm/build $ make
[  0%] Building C object src/utilities/CMakeFiles/utilities-c.dir/upm_utilities.c.o
[  0%] Linking C shared library libupmc-utilities.so
[  0%] Built target utilities-c
[  1%] Building C object src/a110x/CMakeFiles/a110x-c.dir/a110x.c.o
[  1%] Linking C shared library libupmc-a110x.so
[  1%] Built target a110x-c
[  1%] Building CXX object src/a110x/CMakeFiles/a110x.dir/a110x.cxx.o
*** Error in `/usr/bin/c++': double free or corruption (top): 0x005e24b8 ***
Aborted
src/a110x/CMakeFiles/a110x.dir/build.make:62: recipe for target 'src/a110x/CMakeFiles/a110x.dir/a110x.cxx.o' failed
make[2]: *** [src/a110x/CMakeFiles/a110x.dir/a110x.cxx.o] Error 134
CMakeFiles/Makefile2:743: recipe for target 'src/a110x/CMakeFiles/a110x.dir/all' failed
make[1]: *** [src/a110x/CMakeFiles/a110x.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
`

Propanu, 2017-02-13:
Hi @efbick, try specifying your architecture when generating the makefiles with cmake like this:
`cmake -DCMAKE_CXX_FLAGS:STRING=-march=armv6 ..`

You don't have to worry about missing modules, the sensor modules that have a dependency on them simply won't get built.

efbick, 2017-02-16:
Thanks that seemed to work for installing. I attempted to run one of the examples for the mpu9250 and it can't seem to find it. Not sure what happened. Attached the error below.

`Traceback (most recent call last):
  File "/home/pi/Documents/Source/upm/examples/python/mpu9250.py", line 26, in <module>
    from upm import pyupm_mpu9150 as sensorObj
  File "/usr/local/lib/python2.7/dist-packages/upm/pyupm_mpu9150.py", line 17, in <module>
    _pyupm_mpu9150 = swig_import_helper()
  File "/usr/local/lib/python2.7/dist-packages/upm/pyupm_mpu9150.py", line 16, in swig_import_helper
    return importlib.import_module('_pyupm_mpu9150')
  File "/usr/lib/python2.7/importlib/__init__.py", line 37, in import_module
    __import__(name)
ImportError: No module named _pyupm_mpu9150`

Propanu, 2017-02-22:
Hi @efbick. You can run `find /usr -name "_pyupm_mpu9150*"` to check if the .so files have been installed in a different location. There's a chance they were placed under `site-packages` instead of `dist-packages` as expected on your system.