eclipse-upm/upm #525: Intel XDK template of  MCP9808 temperature sensor doesn't work with the Galileo Gen 2

pache1211, 2017-02-27:
Hi I got these errors while running the  MCP9808 template with the Galileo Gen 2:

MRAA Version: v1.5.1
ERROR: /home/root/.node_app_slot/server.js:19
ERROR: var temp = new mcp.MCP9808(6);
  ^
 
Error: Illegal arguments for construction of _exports_MCP9808
  at Error (native)
  at Object.<anonymous> (/home/root/.node_app_slot/server.js:19:12)
  at Module._compile (module.js:409:26)
  at Object.Module._extensions..js (module.js:416:10)
  at Module.load (module.js:343:32)
  at Function.Module._load (module.js:300:12)
  at Function.Module.runMain (module.js:441:10)
  at startup (node.js:139:18)
  at node.js:968:3

Propanu, 2017-02-28:
Hi @pache1211, the default (and only) I2C bus on the Galileo boards is bus 0. Thus your code needs to be changed to `var temp = new mcp.MCP9808(0);`.

pache1211, 2017-02-28:
Hi got the same error:

ERROR: /home/root/.node_app_slot/main.js:33
ERROR: var temp = new mcp.MCP9808(0); 
           ^

Error: Illegal arguments for construction of _exports_MCP9808
    at Error (native)
    at Object.<anonymous> (/home/root/.node_app_slot/main.js:33:12)
    at Module._compile (module.js:409:26)
    at Object.Module._extensions..js (module.js:416:10)
    at Module.load (module.js:343:32)
    at Function.Module._load (module.js:300:12)
    at Function.Module.runMain (module.js:441:10)
    at startup (node.js:139:18)
    at node.js:968:3

Propanu, 2017-03-01:
Check with the vendor for your MCP9808 to find out the I2C address of the sensor. It's configurable through 3 address pins, and we assume a default of 0x18 (A2, A1, A0 grounded). If that's different in your case, say, you have the least significant A0 connected to VCC giving you address 0x19, you'll have to specify it when calling the constructor like so: `var temp = new mcp.MCP9808(0, 0x19);`

pache1211, 2017-03-01:
Thanks! I will try your suggestion.