eclipse-upm/upm #527: Mraa doesn't work properly.

shigosen, 2017-03-04:
Recently I installed mraa library in Edison kit for arduino.Then I try to identify version of this. But mraa doesn't work properly.Following is output.
 
>var m =require('mraa');
Error: Cannot find module 'mraa'
      at Function.Module._resolveFilename  (module.js:325:15)
      at Function.Module._load  (module.js:276:25)
      at Module.require   (module.js:353:17)
      at require  (internal/module.js:12:17)
      at repl:1:8
      at REPLServer.defaultEval  (repl.js:262:27)
      at  bound  (domain.js:287:14)
      at  REPLServer.runBound [as eval]  (domain.js:300:12)
      at  REPLServer.<anonymous >  (repl.js:431:12)
      at emit0ne  (events.js:82:20)

shigosen, 2017-03-04:
Does anyone help me?

Propanu, 2017-03-06:
Set your `NODE_PATH` variable to the location where you installed MRAA. For example:
`export NODE_PATH=/usr/lib/node_modules`.

shigosen, 2017-03-07:
I forgot where Mraa is saved.Could you tell me how do I search the directory or the file is saved Mraa.

Propanu, 2017-03-07:
You can use
```sh
find / -name "mraa\.node" 2>/dev/null
```
This will also filter permission denied error messages.

shigosen, 2017-03-10:
I used command line mentioned above ,but Tera term didn't show any characters.

shigosen, 2017-03-10:
Tera term didn't show any characters but root@username.

shigosen, 2017-03-10:
I don't know what variables should set to the location I installed Mraa.And I couldn't understand command line mentioned above. I assume `export NODE_PATH=/user/lib/node_modules` means set NODE_PATH variable to /user/lib/node_modules directory.

Propanu, 2017-03-15:
Yes, when you set that environment variable, Node.js will use it to search for additional modules at that location. The `find` command mentioned above should give an output similar to this, on an Intel Edison with the latest image:
```sh
root@pedison:~# find / -name "mraa\.node" 2>/dev/null
/usr/lib/node_modules/mraa/mraa.node
```

shigosen, 2017-03-18:
Thanks.I'll try again.